gbionics/idyntree #326: IPOpt dependency

diegoferigo, 2017-06-23:
After #300 has been merged, the build fails if IPOpt is not installed. The CMakeLists should be updated.

```
[ 82%] Building CXX object src/inverse-kinematics/CMakeFiles/idyntree-inverse-kinematics.dir/src/InverseKinematics.cpp.o
In file included from /home/dferigo/git/idyntree/src/inverse-kinematics/src/InverseKinematics.cpp:10:0:
/home/dferigo/git/idyntree/src/inverse-kinematics/include/private/InverseKinematicsData.h:20:34: fatal error: IpIpoptApplication.hpp: No such file or directory
compilation terminated.
src/inverse-kinematics/CMakeFiles/idyntree-inverse-kinematics.dir/build.make:86: recipe for target 'src/inverse-kinematics/CMakeFiles/idyntree-inverse-kinematics.dir/src/InverseKinematics.cpp.o' failed
make[2]: *** [src/inverse-kinematics/CMakeFiles/idyntree-inverse-kinematics.dir/src/InverseKinematics.cpp.o] Error 1
CMakeFiles/Makefile2:550: recipe for target 'src/inverse-kinematics/CMakeFiles/idyntree-inverse-kinematics.dir/all' failed
make[1]: *** [src/inverse-kinematics/CMakeFiles/idyntree-inverse-kinematics.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
```

cc @traversaro @S-Dafarra

traversaro, 2017-06-23:
Mhhh. There was no change of the CMake files in that PR, and iDynTree without using IPOPT compiles fine in iDynTree and AppVeyor. What is the value of `IDYNTREE_USES_IPOPT` on your machine?

diegoferigo, 2017-06-23:
Without `coinor-libipopt-dev`, it is `OFF` and the build fails.

traversaro, 2017-06-23:
The `inverse-dynamics` part should not be compiled if `IDYNTREE_USES_IPOPT` is OFF, see https://github.com/robotology/idyntree/blob/devel/src/CMakeLists.txt#L15 . Did you run cmake?

diegoferigo, 2017-06-23:
PR #327 should fix this

traversaro, 2017-06-29:
Fixed by https://github.com/robotology/idyntree/pull/327 .

closed 2017-06-29 by traversaro