gbionics/idyntree #328: CMake warnings with Qt5 charts

pattacini, 2017-06-23:
Hi,

I'm having the following warnings in CMake while configuring `iDynTree` against all the recent binaries.

```
CMake Warning (dev) at src/tools/idyntree-sole-gui/CMakeLists.txt:16 (message):
  CMAKE_MINIMUM_REQUIRED_VERSION = 2.8.11.  You can remove this.
Call Stack (most recent call first):
  src/tools/idyntree-sole-gui/src/CMakeLists.txt:35 (qticub_deprecate_with_cmake_version)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning at C:/dev/Qt-5.6.1_v14_x86/lib/cmake/Qt5/Qt5Config.cmake:26 (find_package):
  Could not find a package configuration file provided by "Qt5Charts" with
  any of the following names:

    Qt5ChartsConfig.cmake
    qt5charts-config.cmake

  Add the installation prefix of "Qt5Charts" to CMAKE_PREFIX_PATH or set
  "Qt5Charts_DIR" to a directory containing one of the above files.  If
  "Qt5Charts" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  src/tools/plotter/CMakeLists.txt:5 (find_package)

CMake Warning at src/tools/plotter/CMakeLists.txt:5 (find_package):
  Found package configuration file:

    C:/dev/Qt-5.6.1_v14_x86/lib/cmake/Qt5/Qt5Config.cmake

  but it set Qt5_FOUND to FALSE so package "Qt5" is considered to be NOT
  FOUND.  Reason given by package:

  Failed to find Qt5 component "Charts" config file at
  "C:/dev/Qt-5.6.1_v14_x86/lib/cmake/Qt5Charts/Qt5ChartsConfig.cmake"
```

My system is a Win10 with msvc14, compiling in x86 mode.

cc @francesco-romano @traversaro

francesco-romano, 2017-06-23:
Yes, if Qt5 Charts are found, then the iDynTree plotter is compiled.
Unfortunately I don't know if there is a way to make it cleaner during cmake. @traversaro ?

Btw: i think Qt Charts are in Qt >= 5.6

pattacini, 2017-06-23:
Thank @francesco-romano for the explanation.
Yep, it'd be great to make the warning more silent, if possible.

traversaro, 2017-06-23:
I think there are some flags to silence the warning, I will check.

traversaro, 2017-06-29:
Fixed by https://github.com/robotology/idyntree/issues/330 .

closed 2017-06-29 by traversaro

pattacini, 2017-06-29:
Awesome, thanks!