gbionics/idyntree #329: Configuration error in Ubuntu 16.04 if libblas-dev or liblapack-dev are not installed but ipopt it is

traversaro, 2017-06-26:
~~~
-- Could NOT find Doxygen (missing:  DOXYGEN_EXECUTABLE) 
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
blas_PATH
    linked by target "idyntree-inverse-kinematics" in directory /home/isaac/src/isaac-superbuild/isaac/iDynTree/src/inverse-kinematics
    linked by target "idyntree-inverse-kinematics" in directory /home/isaac/src/isaac-superbuild/isaac/iDynTree/src/inverse-kinematics
    linked by target "idyntree-inverse-kinematics" in directory /home/isaac/src/isaac-superbuild/isaac/iDynTree/src/inverse-kinematics
gfortran_PATH
    linked by target "idyntree-inverse-kinematics" in directory /home/isaac/src/isaac-superbuild/isaac/iDynTree/src/inverse-kinematics
lapack_PATH
    linked by target "idyntree-inverse-kinematics" in directory /home/isaac/src/isaac-superbuild/isaac/iDynTree/src/inverse-kinematics

-- Configuring incomplete, errors occurred!
See also "/home/isaac/src/isaac-superbuild/build/isaac/iDynTree/CMakeFiles/CMakeOutput.log".
CMakeFiles/iDynTree.dir/build.make:102: recipe for target 'isaac/iDynTree/CMakeFiles/YCMStamp/iDynTree-configure' failed
~~~

francesco-romano, 2017-06-26:
then IPOpt does not have these dependencies on linux?

traversaro, 2017-06-26:
I think it is related to https://bugs.launchpad.net/ubuntu/+source/coinor-ipopt/+bug/1659835 . I do not have clear if it is actually possible to link to ipopt without linking those libraries.

francesco-romano, 2017-06-26:
@traversaro yes.. that is what I thought.

traversaro, 2017-06-27:
Related issue: https://github.com/robotology/icub-main/issues/418 .
Interestingly, the Debian package `coinor-libipopt-dev` indeed depends on `libblas3` and `liblapack3` packages. However this packages do not install the libraries ` /usr/lib/libblas.so` and `/usr/lib/liblapack.so`, but rather they install  `/usr/lib/blas/libblas.so` and `/usr/lib/lapack/liblapack.so`.
The symlink in `/usr/lib/libblas.so` and `/usr/lib/liblapack.so` is only obtained by installing the `libblas-dev` and `liblapack-dev`:
~~~
Setting up libblas-dev (3.6.0-2ubuntu2) ...
update-alternatives: using /usr/lib/libblas/libblas.so to provide /usr/lib/libblas.so (libblas.so) in auto mode
Setting up liblapack-dev (3.6.0-2ubuntu2) ...
update-alternatives: using /usr/lib/lapack/liblapack.so to provide /usr/lib/liblapack.so (liblapack.so) in auto mode
~~~
However, as ipopt links this libraries undirectly, using ipopt without linking them works fine (confirming that the core problem is that the pkg-config file is malformed). For this reason I think it make sense to merge https://github.com/robotology/idyntree/issues/331 .

traversaro, 2020-11-03:
Fixed by https://github.com/robotology/idyntree/pull/331 .

closed 2020-11-03 by traversaro