gbionics/idyntree #344: Travis failing on devel

traversaro, 2017-08-08:
Since yesterday the Linux travis build on devel failed (and with them, also all related PRs).
The error seems to be this one:
~~~
In file included from /usr/include/octave-4.2.1/octave/mex.h:51:
In file included from /usr/include/octave-4.2.1/octave/octave-config.h:64:
In file included from /usr/lib/gcc/x86_64-linux-gnu/4.8/../../../../include/c++/4.8/cstdint:35:
[1m/usr/lib/gcc/x86_64-linux-gnu/4.8/../../../../include/c++/4.8/bits/c++0x_warning.h:32:2: [0m[0;1;31merror: [0m[1m
      This file requires compiler and library support for the ISO C++ 2011
      standard. This support is currently experimental, and must be enabled with
      the -std=c++11 or -std=gnu++11 compiler options.[0m
#error This file requires compiler and library support for the \
~~~
I guess they updated to the latest octave in the Travis Trusty machine, and the build system is not handling correctly the combination Octave 4.2 + Ubuntu 14.04 .

traversaro, 2017-08-08:
cc @francesco-romano @S-Dafarra at the moment the IK PR is failing for this

S-Dafarra, 2017-08-09:
It seems still happening.

traversaro, 2017-08-09:
The `devel` branch seems to be working fine: https://travis-ci.org/robotology/idyntree/builds/262647307 . If some other branch is still failing, probably it needs to be rebased/merged with the latest devel.

S-Dafarra, 2017-08-09:
Sorry I was mislead by the fact the only one travis check over two was failing.

closed 2017-08-22 by traversaro