gbionics/idyntree #362: Find a smart way to update online the (inverse) covariance of the measurement prior

traversaro, 2017-08-30:
In `BerdySparseMAPSolver` the (inverse) covariance of the measurements is stored as a `iDynTree::SparseMatrix<iDynTree::ColumnMajor>` class.

For using this class for online estimation on the iCub robot of joint torques and external wrenches (scenario similar to the one for which a test was added in https://github.com/robotology/idyntree/pull/359), we need to be able to quickly change online the inverse covariance of the measurement prior, depending on the output of the skin. 

The rational is the following: as we are including all the external force/torque as dynamic variable, we need a way to specify to the algorithm that we know that the external force/torque on most the links is actually zero. We do so by introducing a "fake" sensor (in Berdy sense) measuring the external force/torque on each link, and we assume that all this "fake" sensors measure zero force.

If the skin indicates that there is no external contact on a given link, then we should impose that the external force/torque on that is close to zero by reducing the covariance of the measurement associated to the external force/torque on that link. If on the other hand the skin indicate that there is a contact on that link, but the relative contact force/torque is unknown, the covariance of the "measurement" associated with the external wrenches should be big, to let the estimator estimate the value of the external force. 

On a typical iCub estimation scenario problem, the current performance hit of building the covariance matrices every time is the following: 
~~~
47: iCubTorqueEstimationIntegrationTest working fine for model icub_skin_frames.urdf
47: Number Of Dynamics Variables                 728
47: Number Of Dynamics Equations                 462
47: Number Of Measurements                       308
47: Time to update dynamics variables prior      0.003155
47: Time to update dynamics equations prior      0.001383
47: Time to update measurements prior            0.000669
47: Time for deterministic estimation            0.000182
47: Time for probabilistic estimation            0.002302
~~~ 

Note that *Time for probabilistic estimation* does not include the *Time to update measurements prior* . As the performance hit would be only 0.6 ms on a total estimation time of 2.9 (0.6+2.3) ms on a powerful machine, we can neglect this problem, but on the long run it would be nice to address this. A possible solution is to model all the priors (measurements, dynamical variables and dynamical equations) as diagonal matrices (because until now we actually always set them to be diagonal matrices). In that case the update of the inverse covariance will be reduced to the problem of setting element in a vector. Doing this correctly (using `Eigen::Map<Eigen::DiagonalMatrix<double, Eigen::Dynamic>`) could actually also improve the computational efficiency of the estimation in general.

traversaro, 2017-08-30:
cc @francesco-romano @diegoferigo @claudia-lat @lucaTagliapietra

francesco-romano, 2017-08-30:
Switching to diagonal matrices leads to a "big" change, in the sense that we cannot switch back to full matrix. 
If we think that covariance between measurement variables are negligible, then this can be a choice (and I think the most performance-wise choice).
Otherwise, even accessing single elements of the covariance matrix, which should be possible at this stage, is not a constant operation.

traversaro, 2017-08-30:
> Switching to diagonal matrices leads to a "big" change, in the sense that we cannot switch back to full matrix.

In theory we can keep both, but the interface will be awkward (methods to specify if you want to use the diagonal covariance or the full one). 

> Otherwise, even accessing single elements of the *inverse* covariance matrix, which should be possible at this stage, is not a constant operation

Indeed, I was confused about the const getter of the *inverse* and the implementation of `SparseMatrix::operator(int,int)`, but indeed this is an easy check to do.

francesco-romano, 2017-08-30:
There should be also the non-cost version of the inverse getter.  In this way it is possible to modify single elements. 

Yeah.. I am a bit afraid of the "awkward" interface and how to properly switch between variables in the solver.

traversaro, 2017-09-01:
Using `SparseMatrix::operator(int,int)` is indeed fast enough: 
~~~
47: iCubTorqueEstimationIntegrationTest working fine for model icub_skin_frames.urdf
47: Number Of Dynamics Variables                 728
47: Number Of Dynamics Equations                 462
47: Number Of Measurements                       308
47: Time to update dynamics variables prior      0.002857
47: Time to update dynamics equations prior      0.001221
47: Time to update measurements prior            7e-06
47: Time for deterministic estimation            0.000166
47: Time for probabilistic estimation            0.001805
~~~
I think we can close for now.

closed 2017-09-01 by traversaro