gbionics/idyntree #366: iDynTree::KinDynComputations getFrameBiasAcc() returning nonsense with FrameVelocityRepresentation = MIXED_REPRESENTATION

paulinejmaurice, 2017-09-05:
When calling KinDynComputations::getFrameBiasAcc(index), the last 3 elements (i.e. angular part) of the returned vector appeared to be nonsense (e.g. 10^230), when the FrameVelocityRepresentation is set to MIXED_REPRESENTATION.

I looked in the code of KinDynComputations.cpp, in the function "convertBodyFixedAccelerationToMixedAcceleration" which is called only when MIXED_REPRESENTATION is on. It seems that "mixedAcceleration" is never initialized, which might explain why I get nonsense angular part in the output vector.

traversaro, 2017-09-05:
Thanks for reporting this issue. I was able to reproduce it, I will come back to you shortly.

traversaro, 2017-09-06:
The issue has been fixed in https://github.com/robotology/idyntree/pull/368 . As mentioned in the same PR, there is still an issue in the value returned by  `getFrameBiasAcc` when the representation is `MIXED_REPRESENTATION`. This bug will be fixed soon (please follow https://github.com/robotology/idyntree/issues/370 to be up to date) but it should be negligible for any real world use. 

Thanks a lot for reporting the problem.

closed 2017-09-06 by traversaro