gbionics/idyntree #370: iDynTree::KinDynComputations getFrameBiasAcc() returning slightly wrong data with FrameVelocityRepresentation = MIXED_REPRESENTATION

traversaro, 2017-09-06:
See https://github.com/robotology/idyntree/pull/368 .

ahoarau, 2018-09-12:
@traversaro The default representation is mixed representation is assume ?

traversaro, 2018-09-12:
Yes, anyhow there was actually a much critical bug, see https://github.com/robotology/idyntree/pull/482 .

ahoarau, 2018-09-12:
Anyway to backport to iDynTree version < c++14 ?

traversaro, 2018-09-12:
Agh, I can work on that. Which version is the latest < C++14?

ahoarau, 2018-09-12:
I have 0.8.1

traversaro, 2018-09-12:
Just to understand, which OS+Compiler are you targeting? Are you planning to eventually move to some newer version or is some external constraint, i.e. robot with a fixed version of an OS?

ahoarau, 2018-09-12:
I'm just being lazy/scared to try the new version. 
I had to make a few adjustments to embed it : 
* This annoying warning : https://github.com/syroco/orca/commit/40dd0937ab266495c6f828f3de4ca9fbd3836689
(this was a trick for gazebo)

* These warnings https://github.com/syroco/orca/commit/ff72994a7c2593f51e239c634a2ab070e3c72324

* These warnings https://github.com/syroco/orca/commit/527e9900d9f29b911a856f3c910ad4260322eca4

* These cmake warnings 
```
if (POLICY CMP0048)
    # cmake warns if loaded from a min-3.0-required parent dir, so silence the warning:
    cmake_policy(SET CMP0048 NEW)
endif()
```

* This important catkin support  (discussed in some issue)
https://github.com/syroco/orca/commit/d7b51bf0b9dffacbd1deeb168c33bda304e9fab2#diff-50688af0eb575dc4d216c30f6d423c12

* Quiet ICUB not found https://github.com/syroco/orca/blob/master/external/idyntree-0.8.1/cmake/iDynTreeDependencies.cmake

* The uninstall target should verify that another one is present.

traversaro, 2018-09-12:
Fix cherry-picked on the top of `v0.8.1`: https://github.com/robotology/idyntree/tree/v0.8.1_fixes (don't worry for the Travis failure, it is related to the YARP part). We will try to handle your pan points so you can eventually use a modern version of iDynTree.

ahoarau, 2018-09-12:
@traversaro I owe you many birras. Thanks Silvio !

traversaro, 2018-10-03:
Fixed by https://github.com/traversaro/idyntree/commit/5eec4d4d877b9aabf4dde0f8c6ca78aa8582ba00 .

closed 2018-10-03 by traversaro