gbionics/idyntree #372: Unable to use matlab bindings

fjandrad, 2017-09-08:
I just recently updated and compiled everything.
Yarp in devel, icub-main in traversaros/forceSkinRefactor , iDyntree in useUnkownsUnkownContact and codyco-modules in useForceEstimatedFromSkin.

It all compiles without problems but when I try to launch a matlab script using iDyntree classes I get the following:
Invalid MEX-file '/home/fandradechavez/dev/codyco-superbuild/build/install/mex/iDynTreeMEX.mexa64': Missing dependent
shared libraries:
'libidyntree-high-level-kdl.so' required by
'/home/fandradechavez/dev/codyco-superbuild/build/install/mex/iDynTreeMEX.mexa64'
'libidyntree-visualization.so' required by
'/home/fandradechavez/dev/codyco-superbuild/build/install/mex/iDynTreeMEX.mexa64'
'libidyntree-high-level.so' required by
'/home/fandradechavez/dev/codyco-superbuild/build/install/mex/iDynTreeMEX.mexa64'
'libidyntree-estimation.so' required by
'/home/fandradechavez/dev/codyco-superbuild/build/install/mex/iDynTreeMEX.mexa64'
'libidyntree-regressors.so' required by
'/home/fandradechavez/dev/codyco-superbuild/build/install/mex/iDynTreeMEX.mexa64'
'libidyntree-modelio-urdf.so' required by
'/home/fandradechavez/dev/codyco-superbuild/build/install/mex/iDynTreeMEX.mexa64'
'libidyntree-sensors.so' required by '/home/fandradechavez/dev/codyco-superbuild/build/install/mex/iDynTreeMEX.mexa64'
'libidyntree-model.so' required by '/home/fandradechavez/dev/codyco-superbuild/build/install/mex/iDynTreeMEX.mexa64'
'libidyntree-core.so' required by '/home/fandradechavez/dev/codyco-superbuild/build/install/mex/iDynTreeMEX.mexa64'

Missing symbol '_ZdlPvm' required by '/home/fandradechavez/dev/codyco-superbuild/build/install/mex/iDynTreeMEX.mexa64'
Missing symbol '_ZN8iDynTree10IndexRange12InvalidRangeEv' required by
'/home/fandradechavez/dev/codyco-superbuild/build/install/mex/iDynTreeMEX.mexa64'
Missing symbol
'_ZN8iDynTree10Regressors26DynamicsRegressorGenerator13setRobotStateERKNS_13VectorDynSizeES4_S4_RKNS_5TwistE' required
by '/home/fandradechavez/dev/codyco-superbuild/build/install/mex/iDynTreeMEX.mexa64'
Missing symbol
'_ZN8iDynTree10Regressors26DynamicsRegressorGenerator13setRobotStateERKNS_13VectorDynSizeES4_S4_RKNS_9TransformERKNS_5TwistESA_SA_'
required by '/home/fandradechavez/dev/codyco-superbuild/build/install/mex/iDynTreeMEX.mexa64'
Missing symbol '_ZN8iDynTree10Regressors26DynamicsRegressorGenerator15getBaseLinkNameB5cxx11Ev' required by
'/home/fandradechavez/dev/codyco-superbuild/build/install/mex/iDynTreeMEX.mexa64'
.
.
.
.
and many many more like this until 
Error in iDynTree.ExtWrenchesAndJointTorquesEstimator (line 12)
        tmp = iDynTreeMEX(1390, varargin{:});

Error in readExperiment (line 36)
      estimator = iDynTree.ExtWrenchesAndJointTorquesEstimator();

I do not know if its a problem from the matlab bindings alone but it reminds me of the error @francesco-romano  help solve. ( the missing symbol problem ) I remember he said he found a temporal solution, but I could not find it in any of the issues did you put it somewhere?
 Any ideas @traversaro ?

traversaro, 2017-09-09:
It seems there could be a problem a RPATH problem, probably connected to the use of Ubuntu 17.04 . 
Can you try to add `/home/fandradechavez/dev/codyco-superbuild/build/install/lib` to the [`LD_LIBRARY_PATH`](http://tldp.org/HOWTO/Program-Library-HOWTO/shared-libraries.html) enviromental variable? 

Are you launching matlab from the terminal?

fjandrad, 2017-09-09:
Yes I'm launching from terminal I had the path on the LD_LIBRARY_PATH but it was incomplete missing the /fandradechavez part. Now the shared libraries error is not there anymore but all the Missing symbol are still there.

What else could it be?

my cmake version is 3.7.2 if it helps at all
g++ (Ubuntu 6.3.0-12ubuntu2) 6.3.0 20170406
do not know what else info could be useful

fjandrad, 2017-09-09:
Apparently Matlab 2017a uses gcc 4.9 so it has a problem with the g++ i have. 

adding 
`export LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libstdc++.so.6:/usr/lib/x86_64-linux-gnu/libprotobuf.so.9`

to the bashrc seems to solve the problem

traversaro, 2017-09-09:
I see, a different instance of the usual https://github.com/robotology/codyco-superbuild/issues/141 . 
By the way, I never considered the problem, but I wonder if being the ABI of GCC 4.9 and GCC >= 5 incompatible will be creating troubles at the Matlab bindings level.

traversaro, 2017-10-10:
Closing as it is a well known problem at the Matlab level.

closed 2017-10-10 by traversaro

GiulioRomualdi, 2019-01-29:
I had the same problem on my computer (Ubuntu 18.04 and Matlab 2017b). I fixed the issue by adding `export LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libstdc++.so.6` in the `bashrc` 

Thanks @fjandrad :smile: 
cc @gabrielenava

traversaro, 2019-01-29:
This needs to be properly documented at robotology-superbuild level (or at least in place where it can be referred by all projects), see https://github.com/robotology/robotology-superbuild/issues/64 for more details.