gbionics/idyntree #375: Cannot compile bindings on macOS

francesco-romano, 2017-09-25:
The error is the following

```
/Users/makaveli/Projects/src/codyco-superbuild/libraries/iDynTree/bindings/matlab/autogenerated/iDynTreeMATLAB_wrap.cxx:4347:9: error: no member named 'isfinite' in namespace 'std'; did you mean 'finite'?
    if (SWIG_Float_Overflow_Check(v)) {
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/makaveli/Projects/src/codyco-superbuild/libraries/iDynTree/bindings/matlab/autogenerated/iDynTreeMATLAB_wrap.cxx:4335:73: note: expanded from macro 'SWIG_Float_Overflow_Check'
# define SWIG_Float_Overflow_Check(X) ((X < -FLT_MAX || X > FLT_MAX) && SWIG_isfinite(X))
                                                                        ^~~~~~~~~~~~~~~~
/Users/makaveli/Projects/src/codyco-superbuild/libraries/iDynTree/bindings/matlab/autogenerated/iDynTreeMATLAB_wrap.cxx:4323:29: note: expanded from macro 'SWIG_isfinite'
#  define SWIG_isfinite(X) (std::isfinite(X))
                            ^~~~~
In file included from /Users/makaveli/Projects/src/codyco-superbuild/libraries/iDynTree/bindings/matlab/autogenerated/iDynTreeMATLAB_wrap.cxx:173:
In file included from /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/math.h:301:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.13.sdk/usr/include/math.h:757:12: note: 'finite' declared here
extern int finite(double) __OSX_AVAILABLE_BUT_DEPRECATED(__MAC_10_0, __MAC_10_9, __IPHONE_NA, __IPHONE_NA);
           ^
1 error generated.

```
Note the SDK is 10.13, i.e. the new macOS version

traversaro, 2017-09-25:
It is probably a SWIG upstream issue. I think a possible workaround is to add an `#include <cmath>` in https://github.com/robotology/idyntree/blob/devel/bindings/iDynTree.i#L29 .

francesco-romano, 2017-09-25:
Solved thanks!

closed 2017-09-25 by francesco-romano

traversaro, 2017-09-25:
Related SWIG issue: https://github.com/swig/swig/issues/615 .