gbionics/idyntree #385: Timing statistics of IK

francesco-romano, 2017-10-25:
In this issue we collect timing statistics regarding the inverse kinematics with the aim of finding (and removing) potential bottlenecks.

The IK is configured in the following way:
- Robot: iCub
- Eq. Constraints: left foot (full 3D : dof=6), right foot (full 3D : dof=6), CoM position (: dof=3)
- In. Constraints: none
- Joint limits
- Parametrization: RPY
- Cost: Joints regularisation, Torso orientation
- Controlled joints: 15 (legs and torso)

In general IK returns in less than 10 iterations, a part some sporadic cases.

Here is a very long run (longest)
```
Number of Iterations....: 133

                                   (scaled)                 (unscaled)
Objective...............:   1.4621712534352573e-03    1.4621712534352573e-03
Dual infeasibility......:   5.5420759928266284e-05    5.5420759928266284e-05
Constraint violation....:   1.9349939706092556e-09    1.9349939706092556e-09
Complementarity.........:   2.5659855140747359e-10    2.5659855140747359e-10
Overall NLP error.......:   5.5420759928266284e-05    5.5420759928266284e-05

Number of objective function evaluations             = 525
Number of objective gradient evaluations             = 128
Number of equality constraint evaluations            = 525
Number of inequality constraint evaluations          = 525
Number of equality constraint Jacobian evaluations   = 135
Number of inequality constraint Jacobian evaluations = 135
Number of Lagrangian Hessian evaluations             = 0
Total CPU secs in IPOPT (w/o function evaluations)   =      0.068
Total CPU secs in NLP function evaluations           =      0.040

Timing Statistics:

OverallAlgorithm....................:      0.108 (sys:      0.032 wall:      0.134)
 PrintProblemStatistics.............:      0.000 (sys:      0.000 wall:      0.000)
 InitializeIterates.................:      0.000 (sys:      0.000 wall:      0.000)
 UpdateHessian......................:      0.004 (sys:      0.000 wall:      0.003)
 OutputIteration....................:      0.000 (sys:      0.000 wall:      0.001)
 UpdateBarrierParameter.............:      0.012 (sys:      0.012 wall:      0.013)
 ComputeSearchDirection.............:      0.028 (sys:      0.012 wall:      0.037)
 ComputeAcceptableTrialPoint........:      0.044 (sys:      0.008 wall:      0.069)
 AcceptTrialPoint...................:      0.000 (sys:      0.000 wall:      0.003)
 CheckConvergence...................:      0.016 (sys:      0.000 wall:      0.007)
PDSystemSolverTotal.................:      0.032 (sys:      0.016 wall:      0.049)
 PDSystemSolverSolveOnce............:      0.028 (sys:      0.016 wall:      0.042)
 ComputeResiduals...................:      0.000 (sys:      0.000 wall:      0.003)
 StdAugSystemSolverMultiSolve.......:      0.012 (sys:      0.008 wall:      0.018)
 LinearSystemScaling................:      0.000 (sys:      0.000 wall:      0.000)
 LinearSystemSymbolicFactorization..:      0.000 (sys:      0.000 wall:      0.000)
 LinearSystemFactorization..........:      0.000 (sys:      0.000 wall:      0.004)
 LinearSystemBackSolve..............:      0.004 (sys:      0.000 wall:      0.004)
 LinearSystemStructureConverter.....:      0.000 (sys:      0.000 wall:      0.000)
  LinearSystemStructureConverterInit:      0.000 (sys:      0.000 wall:      0.000)
QualityFunctionSearch...............:      0.000 (sys:      0.000 wall:      0.005)
TryCorrector........................:      0.000 (sys:      0.000 wall:      0.000)
Task1...............................:      0.000 (sys:      0.000 wall:      0.001)
Task2...............................:      0.000 (sys:      0.000 wall:      0.001)
Task3...............................:      0.000 (sys:      0.000 wall:      0.000)
Task4...............................:      0.000 (sys:      0.000 wall:      0.000)
Task5...............................:      0.000 (sys:      0.000 wall:      0.001)
Function Evaluations................:      0.040 (sys:      0.004 wall:      0.042)
 Objective function.................:      0.000 (sys:      0.000 wall:      0.001)
 Objective function gradient........:      0.008 (sys:      0.000 wall:      0.001)
 Equality constraints...............:      0.028 (sys:      0.004 wall:      0.037)
 Inequality constraints.............:      0.004 (sys:      0.000 wall:      0.001)
 Equality constraint Jacobian.......:      0.000 (sys:      0.000 wall:      0.000)
 Inequality constraint Jacobian.....:      0.000 (sys:      0.000 wall:      0.001)
[... 97 lines cut ...]
 PDSystemSolverSolveOnce............:      0.000 (sys:      0.000 wall:      0.001)
 ComputeResiduals...................:      0.000 (sys:      0.000 wall:      0.000)
 StdAugSystemSolverMultiSolve.......:      0.000 (sys:      0.000 wall:      0.001)
 LinearSystemScaling................:      0.000 (sys:      0.000 wall:      0.000)
 LinearSystemSymbolicFactorization..:      0.000 (sys:      0.000 wall:      0.000)
 LinearSystemFactorization..........:      0.000 (sys:      0.000 wall:      0.000)
 LinearSystemBackSolve..............:      0.000 (sys:      0.000 wall:      0.000)
 LinearSystemStructureConverter.....:      0.000 (sys:      0.000 wall:      0.000)
  LinearSystemStructureConverterInit:      0.000 (sys:      0.000 wall:      0.000)
QualityFunctionSearch...............:      0.000 (sys:      0.000 wall:      0.002)
TryCorrector........................:      0.000 (sys:      0.000 wall:      0.000)
Task1...............................:      0.000 (sys:      0.000 wall:      0.000)
Task2...............................:      0.000 (sys:      0.000 wall:      0.000)
Task3...............................:      0.000 (sys:      0.000 wall:      0.000)
Task4...............................:      0.000 (sys:      0.000 wall:      0.000)
Task5...............................:      0.000 (sys:      0.000 wall:      0.000)
Function Evaluations................:      0.000 (sys:      0.000 wall:      0.001)
 Objective function.................:      0.000 (sys:      0.000 wall:      0.000)
 Objective function gradient........:      0.000 (sys:      0.000 wall:      0.000)
 Equality constraints...............:      0.000 (sys:      0.000 wall:      0.001)
 Inequality constraints.............:      0.000 (sys:      0.000 wall:      0.000)
 Equality constraint Jacobian.......:      0.000 (sys:      0.000 wall:      0.000)
 Inequality constraint Jacobian.....:      0.000 (sys:      0.000 wall:      0.000)
 Lagrangian Hessian.................:      0.000 (sys:      0.000 wall:      0.000)
```

francesco-romano, 2017-10-25:
cc @pattacini @S-Dafarra @traversaro @DanielePucci

francesco-romano, 2017-10-25:
I started developing the library by using quaternion parametrisation.
Unfortunately we never tested on the robot this representation, and I remember some complains by the IPOPT derivative checker.

pattacini, 2017-10-25:
@francesco-romano I had my past developments focused on the use of **axis-angle**, for which we have access to good descriptions of the derivative (ref. Sciavicco-Siciliano)

francesco-romano, 2017-10-25:
That's great. We could extend [this enum](https://github.com/robotology/idyntree/blob/master/src/inverse-kinematics/include/iDynTree/InverseKinematics.h#L36) supporting also **axis-angle**

S-Dafarra, 2017-10-25:
Just to provide some data:
- Number of IK iterations: 2799
## Timings
- Sample Mean: 3.9614 ms
- Sample Variance: 0.1205 ms
- Max time: 556 ms
- Min time: <1ms
- Full data plot:
[image]
- Plot between 1000 and 1999 IK iterations:
[image]

## Iterations
- Sample Mean: 5.6963 iterations
- Sample Variance: 141.2587 iterations
- Max number of iterations: 609
- Min number of iterations: 0
- Full data plot:
[image]
- Plot between 1000 and 1999 IK iterations:
[image]

Ipopt was using ``ma27``.

##Data
[data.zip](https://github.com/robotology/idyntree/files/1414987/data.zip)

DanielePucci, 2017-10-25:
@S-Dafarra thanks a lot for the plots! Just a curiosity, which were the characteristics of the machine used for the tests? Probably, it would be nice to write them here as per reference

lrapetti, 2018-11-26:
Regarding this kind of performance analysis, I think it would be to worth to have cpp interface for reading the [Statistics](https://www.coin-or.org/Doxygen/Ipopt/classIpopt_1_1SolveStatistics.html) of the IK? 

For my understanding, at the moment performance statistics (e.g. `Number of Iterations`) can be only retrieved setting high [verbosity](http://wiki.icub.org/codyco/dox/html/idyntree/html/classiDynTree_1_1InverseKinematics.html#a7fd58eaeec333ffe4a8b1cf68f48199a) and parsing the std output.