gbionics/idyntree #390: CMake warnings when not using icub/yarp

ahoarau, 2017-11-10:
Is there any way to add a quiet `find_package` `QUIET` for `iCub` and `YARP` ?
```bash
CMake Warning at /Users/hoarau/dev/idyntree-0.8.1/cmake/iDynTreeDependencies.cmake:38 (find_package):
  By not providing "FindICUB.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "ICUB", but
  CMake did not find one.

  Could not find a package configuration file provided by "ICUB" with any of
  the following names:

    ICUBConfig.cmake
    icub-config.cmake

  Add the installation prefix of "ICUB" to CMAKE_PREFIX_PATH or set
  "ICUB_DIR" to a directory containing one of the above files.  If "ICUB"
  provides a separate development package or SDK, be sure it has been
  installed.
Call Stack (most recent call first):
  external/idyntree-0.8.1/CMakeLists.txt:39 (include)
```

``` 
CMake Warning at /Users/hoarau/dev/idyntree-0.8.1/cmake/iDynTreeDependencies.cmake:14 (find_package):
  By not providing "FindYARP.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "YARP", but
  CMake did not find one.

  Could not find a package configuration file provided by "YARP" (requested
  version 2.3.62) with any of the following names:

    YARPConfig.cmake
    yarp-config.cmake

  Add the installation prefix of "YARP" to CMAKE_PREFIX_PATH or set
  "YARP_DIR" to a directory containing one of the above files.  If "YARP"
  provides a separate development package or SDK, be sure it has been
  installed.
Call Stack (most recent call first):
  external/idyntree-0.8.1/cmake/iDynTreeDependencies.cmake:44 (idyntree_handle_dependency)
  external/idyntree-0.8.1/CMakeLists.txt:39 (include)
```

traversaro, 2017-11-10:
This make totally sense. @ahoarau I am going to comment on the other issues, sorry for the delay. :pray:

traversaro, 2017-11-10:
It should be sufficient to add a QUIET in those points:
* https://github.com/robotology/idyntree/blob/master/cmake/iDynTreeDependencies.cmake#L20
* https://github.com/robotology/idyntree/blob/master/cmake/iDynTreeDependencies.cmake#L22
* https://github.com/robotology/idyntree/blob/master/cmake/iDynTreeDependencies.cmake#L41

ahoarau, 2017-11-10:
> This make totally sense. @ahoarau I am going to comment on the other issues, sorry for the delay.

No problem mister silvio !

ahoarau, 2018-09-16:
@traversaro ping on this !

closed 2018-09-16 by traversaro