gbionics/idyntree #394: Installed MATLAB bindings do not work correctly in macOS Sierra 10.12.6 CMake 3.10.0 and Matlab 2017a

lucaTagliapietra, 2017-11-23:
The error is the following:
~~~
 a = iDynTree.Vector3()
Invalid MEX-file '/Users/ltagliapietra/Documents/research/iit/idyntree/install/mex/iDynTreeMEX.mexmaci64': Missing dependent shared libraries:
'@rpath/libidyntree-estimation.dylib' required by '/Users/ltagliapietra/Documents/research/iit/idyntree/install/mex/iDynTreeMEX.mexmaci64'
'@rpath/libidyntree-high-level.dylib' required by '/Users/ltagliapietra/Documents/research/iit/idyntree/install/mex/iDynTreeMEX.mexmaci64'
'@rpath/libidyntree-high-level-kdl.dylib' required by '/Users/ltagliapietra/Documents/research/iit/idyntree/install/mex/iDynTreeMEX.mexmaci64'
'@rpath/libidyntree-regressors.dylib' required by '/Users/ltagliapietra/Documents/research/iit/idyntree/install/mex/iDynTreeMEX.mexmaci64'
'@rpath/libidyntree-visualization.dylib' required by '/Users/ltagliapietra/Documents/research/iit/idyntree/install/mex/iDynTreeMEX.mexmaci64'
'@rpath/libidyntree-modelio-urdf.dylib' required by '/Users/ltagliapietra/Documents/research/iit/idyntree/install/mex/iDynTreeMEX.mexmaci64'
'@rpath/libidyntree-sensors.dylib' required by '/Users/ltagliapietra/Documents/research/iit/idyntree/install/mex/iDynTreeMEX.mexmaci64'
'@rpath/libidyntree-model.dylib' required by '/Users/ltagliapietra/Documents/research/iit/idyntree/install/mex/iDynTreeMEX.mexmaci64'
'@rpath/libidyntree-core.dylib' required by '/Users/ltagliapietra/Documents/research/iit/idyntree/install/mex/iDynTreeMEX.mexmaci64'

Missing symbol '__Unwind_Resume' required by '/Users/ltagliapietra/Documents/research/iit/idyntree/install/mex/iDynTreeMEX.mexmaci64'
Missing symbol '__ZN8iDynTree10IndexRange12InvalidRangeEv' required by
'/Users/ltagliapietra/Documents/research/iit/idyntree/install/mex/iDynTreeMEX.mexmaci64'
Missing symbol '__ZN8iDynTree10Regressors26DynamicsRegressorGenerator13setRobotStateERKNS_13VectorDynSizeES4_S4_RKNS_5TwistE' required by
'/Users/ltagliapietra/Documents/research/iit/idyntree/install/mex/iDynTreeMEX.mexmaci64'
Missing symbol '__ZN8iDynTree10Regressors26DynamicsRegressorGenerator13setRobotStateERKNS_13VectorDynSizeES4_S4_RKNS_9TransformERKNS_5TwistESA_SA_'
required by '/Users/ltagliapietra/Documents/research/iit/idyntree/install/mex/iDynTreeMEX.mexmaci64'
Missing symbol '__ZN8iDynTree10Regressors26DynamicsRegressorGenerator15getBaseLinkNameEv' required by
'/Users/ltagliapietra/Documents/research/iit/idyntree/install/mex/iDynTreeMEX.mexmaci64'
Missing symbol '__ZN8iDynTree10Regressors26DynamicsRegressorGenerator16computeRegressorERNS_13MatrixDynSizeERNS_13VectorDynSizeE' required by
'/Users/ltagliapietra/Documents/research/iit/idyntree/install/mex/iDynTreeMEX.mexmaci64'
Missing symbol '__ZN8iDynTree10Regressors26DynamicsRegressorGenerator18getModelParametersERNS_13VectorDynSizeE' required by
'/Users/ltagliapietra/Documents/research/iit/idyntree/install/mex/iDynTreeMEX.mexmaci64'
Missing symbol '__ZN8iDynTree10Regressors26DynamicsRegressorGenerator20getDescriptionOfLinkEi' required by
'/Users/ltagliapietra/Documents/research/iit/idyntree/install/mex/iDynTreeMEX.mexmaci64'
Missing symbol '__ZN8iDynTree10Regressors26DynamicsRegressorGenerator21getDescriptionOfLinksEv' required by
'/Users/ltagliapietra/Documents/research/iit/idyntree/install/mex/iDynTreeMEX.mexmaci64'
Missing symbol '__ZN8iDynTree10Regressors26DynamicsRegressorGenerator22getDescriptionOfOutputEi' required by
'/Users/ltagliapietra/Documents/research/iit/idyntree/install/mex/iDynTreeMEX.mexmaci64'
Missing symbol '__ZN8iDynTree10Regressors26DynamicsRegressorGenerator22getSensorsMeasurementsEv' required by
'/Users/ltagliapietra/Documents/research/iit/idyntree/install/mex/iDynTreeMEX.mexmaci64'
Missing symbol '__ZN8iDynTree10Regressors26DynamicsRegressorGenerator23getDescriptionOfOutputsEv' required by
'/Users/ltagliapietra/Documents/research/iit/idyntree/install/mex/iDynTreeMEX.mexmaci64'
Missing symbol '__ZN8iDynTree10Regressors26DynamicsRegressorGenerator25getDescriptionOfParameterEibd' required by
'/Users/ltagliapietra/Documents/research/iit/idyntree/install/mex/iDynTreeMEX.mexmaci64'
Missing symbol '__ZN8iDynTree10Regressors26DynamicsRegressorGenerator26generate_random_regressorsERNS_13MatrixDynSizeEbbi' required by
'/Users/ltagliapietra/Documents/research/iit/idyntree/install/mex/iDynTreeMEX.mexmaci64'
Missing symbol '__ZN8iDynTree10Regressors26DynamicsRegressorGenerator26getDescriptionOfParametersERKNS_13VectorDynSizeE' required by
'/Users/ltagliapietra/Documents/research/iit/idyntree/install/mex/iDynTreeMEX.mexmaci64'
Missing symbol '__ZN8iDynTree10Regressors26DynamicsRegressorGenerator26getDescriptionOfParametersEv' required by
'/Users/ltagliapietra/Documents/research/iit/idyntree/install/mex/iDynTreeMEX.mexmaci64'
Missing symbol '__ZN8iDynTree10Regressors26DynamicsRegressorGenerator26setTorqueSensorMeasurementERNS_13VectorDynSizeE' required by
'/Users/ltagliapietra/Documents/research/iit/idyntree/install/mex/iDynTreeMEX.mexmaci64'
Missing symbol '__ZN8iDynTree10Regressors26DynamicsRegressorGenerator26setTorqueSensorMeasurementEid' required by
'/Users/ltagliapietra/Documents/research/iit/idyntree/install/mex/iDynTreeMEX.mexmaci64'
Missing symbol
'__ZN8iDynTree10Regressors26DynamicsRegressorGenerator30loadRegressorStructureFromFileERKNSt3__112basic_stringIcNS2_11char_traitsIcEENS2_9allocatorIcEEEE'
required by '/Users/ltagliapietra/Documents/research/iit/idyntree/install/mex/iDynTreeMEX.mexmaci64'
Missing symbol '__ZN8iDynTree10Regressors26DynamicsRegressorGenerator31getDescriptionOfDegreeOfFreedomEi' required by
'/Users/ltagliapietra/Documents/research/iit/idyntree/install/mex/iDynTreeMEX.mexmaci64'
Missing symbol '__ZN8iDynTree10Regressors26DynamicsRegressorGenerator32getDescriptionOfDegreesOfFreedomEv' required by
'/Users/ltagliapietra/Documents/research/iit/idyntree/install/mex/iDynTreeMEX.mexmaci64'
Missing symbol
'__ZN8iDynTree10Regressors26DynamicsRegressorGenerator32loadRegressorStructureFromStringERKNSt3__112basic_stringIcNS2_11char_traitsIcEENS2_9allocatorIcEEEE'
required by '/Users/ltagliapietra/Documents/research/iit/idyntree/install/mex/iDynTreeMEX.mexmaci64'
Missing symbol
'__ZN8iDynTree10Regressors26DynamicsRegressorGenerator32loadRobotAndSensorsModelFromFileERKNSt3__112basic_stringIcNS2_11char_traitsIcEENS2_9allocatorIcEEEESA_'
required by '/Users/ltagliapietra/Documents/research/iit/idyntree/install/mex/iDynTreeMEX.mexmaci64'
Missing symbol
'__ZN8iDynTree10Regressors26DynamicsRegressorGenerator34loadRobotAndSensorsModelFromStringERKNSt3__112basic_stringIcNS2_11char_traitsIcEENS2_9allocatorIcEEEESA_'
required by '/Users/ltagliapietra/Documents/research/iit/idyntree/install/mex/iDynTreeMEX.mexmaci64'
Missing symbol '__ZN8iDynTree10Regressors26DynamicsRegressorGenerator36computeFixedBaseIdentifiableSubspaceERNS_13MatrixDynSizeE' required by
'/Users/ltagliapietra/Documents/research/iit/idyntree/install/mex/iDynTreeMEX.mexmaci64'
Missing symbol '__ZN8iDynTree10Regressors26DynamicsRegressorGenerator39computeFloatingBaseIdentifiableSubspaceERNS_13MatrixDynSizeE' required by
'/Users/ltagliapietra/Documents/research/iit/idyntree/install/mex/iDynTreeMEX.mexmaci64'
Missing symbol '__ZN8iDynTree10Regressors26DynamicsRegressorGenerator7isValidEv' required by
[... 1994 lines cut ...]
Missing symbol '__ZNK8iDynTree4Axis10isParallelERKS0_d' required by
'/Users/ltagliapietra/Documents/research/iit/idyntree/install/mex/iDynTreeMEX.mexmaci64'
Missing symbol '__ZNK8iDynTree4Axis12getDirectionEv' required by
'/Users/ltagliapietra/Documents/research/iit/idyntree/install/mex/iDynTreeMEX.mexmaci64'
Missing symbol '__ZNK8iDynTree4Axis16getRotationTwistEd' required by
'/Users/ltagliapietra/Documents/research/iit/idyntree/install/mex/iDynTreeMEX.mexmaci64'
Missing symbol '__ZNK8iDynTree4Axis16reservedToStringEv' required by
'/Users/ltagliapietra/Documents/research/iit/idyntree/install/mex/iDynTreeMEX.mexmaci64'
Missing symbol '__ZNK8iDynTree4Axis19getTranslationTwistEd' required by
'/Users/ltagliapietra/Documents/research/iit/idyntree/install/mex/iDynTreeMEX.mexmaci64'
Missing symbol '__ZNK8iDynTree4Axis20getRotationTransformEd' required by
'/Users/ltagliapietra/Documents/research/iit/idyntree/install/mex/iDynTreeMEX.mexmaci64'
Missing symbol '__ZNK8iDynTree4Axis21getRotationSpatialAccEd' required by
'/Users/ltagliapietra/Documents/research/iit/idyntree/install/mex/iDynTreeMEX.mexmaci64'
Missing symbol '__ZNK8iDynTree4Axis23getTranslationTransformEd' required by
'/Users/ltagliapietra/Documents/research/iit/idyntree/install/mex/iDynTreeMEX.mexmaci64'
[... cut ...]
Missing symbol '__ZNK8iDynTree4Axis33getTranslationTransformDerivativeEd' required by
'/Users/ltagliapietra/Documents/research/iit/idyntree/install/mex/iDynTreeMEX.mexmaci64'
Missing symbol '__ZNK8iDynTree4Axis7reverseEv' required by '/Users/ltagliapietra/Documents/research/iit/idyntree/install/mex/iDynTreeMEX.mexmaci64'
Missing symbol '__ZNK8iDynTree4Axis8toStringEv' required by '/Users/ltagliapietra/Documents/research/iit/idyntree/install/mex/iDynTreeMEX.mexmaci64'
Missing symbol '__ZNK8iDynTree4Axis9getOriginEv'

lucaTagliapietra, 2017-11-23:
The problem is probably RPATH-related, and is not affecting the bindings in the build directory, so a workaround is to use the bindings from the build directory using the following commands:
~~~
addpath('/Users/ltagliapietra/ws_traversaro/idyntree/build/lib');addpath('/Users/ltagliapietra/ws_traversaro/idyntree/bindings/matlab/autogenerated')
~~~
(Note: this directory can be extracted by running `ctest -VV -R matlab`).

traversaro, 2017-11-23:
I was not able to reproduce the problem in Matlab 2016a, both with CMake 3.9 and CMake 3.10 .

DiegoRomeres, 2018-03-21:
I have a similar problem. I installed codyco-superbuild today in my linux machine with Ubuntu 16.04, Matlab 2017a and CMake 3.10.3. I tried to compile this [example](https://github.com/robotology-playground/idyntree/blob/master/examples/matlab/iCubLeftArmDynamicRegressors/iCubLeftArmFTRegressor.m) and I have the following error:

```
>> iCubLeftArmFTRegressor
Invalid MEX-file
'/homes/romeres/Repositories/codyco-superbuild/build/libraries/iDynTree/lib/iDynTreeMEX.mexa64':
Missing symbol '_ZNSt12out_of_rangeC1EPKc' required by
'/homes/romeres/Repositories/codyco-superbuild/build/libraries/iDynTree/lib/iDynTreeMEX.mexa64'
Missing symbol '_ZNSt16invalid_argumentC1EPKc' required by
'/homes/romeres/Repositories/codyco-superbuild/build/libraries/iDynTree/lib/iDynTreeMEX.mexa64'
Missing symbol '_ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE4swapERS4_' required by
'/homes/romeres/Repositories/codyco-superbuild/build/libraries/iDynTree/lib/iDynTreeMEX.mexa64'
Missing symbol '_ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE9_M_appendEPKcm' required by
'/homes/romeres/Repositories/codyco-superbuild/build/libraries/iDynTree/lib/iDynTreeMEX.mexa64'
Missing symbol '_ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE9_M_assignERKS4_' required by
'/homes/romeres/Repositories/codyco-superbuild/build/libraries/iDynTree/lib/iDynTreeMEX.mexa64'
Missing symbol '_ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE9_M_createERmm' required by
'/homes/romeres/Repositories/codyco-superbuild/build/libraries/iDynTree/lib/iDynTreeMEX.mexa64'
Missing symbol '_ZNSt7__cxx1118basic_stringstreamIcSt11char_traitsIcESaIcEED1Ev' required by
'/homes/romeres/Repositories/codyco-superbuild/build/libraries/iDynTree/lib/iDynTreeMEX.mexa64'
Missing symbol '_ZTTNSt7__cxx1118basic_stringstreamIcSt11char_traitsIcESaIcEEE' required by
'/homes/romeres/Repositories/codyco-superbuild/build/libraries/iDynTree/lib/iDynTreeMEX.mexa64'
Missing symbol '_ZTVNSt7__cxx1115basic_stringbufIcSt11char_traitsIcESaIcEEE' required by
'/homes/romeres/Repositories/codyco-superbuild/build/libraries/iDynTree/lib/iDynTreeMEX.mexa64'
Missing symbol '_ZTVNSt7__cxx1118basic_stringstreamIcSt11char_traitsIcESaIcEEE' required by
'/homes/romeres/Repositories/codyco-superbuild/build/libraries/iDynTree/lib/iDynTreeMEX.mexa64'
Missing symbol '_ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE9_M_createERmm' required by
'/homes/romeres/Repositories/codyco-superbuild/build/libraries/iDynTree/lib/libidyntree-high-level-kdl.so->/homes/romeres/Repositories/codyco-superbuild/build/libraries/iDynTree/lib/iDynTreeMEX.mexa64'
Missing symbol '_ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEED1Ev' required by
'/homes/romeres/Repositories/codyco-superbuild/build/libraries/iDynTree/lib/libidyntree-high-level-kdl.so->/homes/romeres/Repositories/codyco-superbuild/build/libraries/iDynTree/lib/iDynTreeMEX.mexa64'
Missing symbol '_ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE4swapERS4_' required by
'/homes/romeres/Repositories/codyco-superbuild/build/libraries/iDynTree/lib/libidyntree-core.so->/homes/romeres/Repositories/codyco-superbuild/build/libraries/iDynTree/lib/libidyntree-high-level-kdl.so->/homes/romeres/Repositories/codyco-superbuild/build/libraries/iDynTree/lib/iDynTreeMEX.mexa64'
Missing symbol '_ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE9_M_assignERKS4_' required by
'/homes/romeres/Repositories/codyco-superbuild/build/libraries/iDynTree/lib/libidyntree-core.so->/homes/romeres/Repositories/codyco-superbuild/build/libraries/iDynTree/lib/libidyntree-high-level-kdl.so->/homes/romeres/Repositories/codyco-superbuild/build/libraries/iDynTree/lib/iDynTreeMEX.mexa64'
Missing symbol '_ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE9_M_createERmm' required by
'/homes/romeres/Repositories/codyco-superbuild/build/libraries/iDynTree/lib/libidyntree-core.so->/homes/romeres/Repositories/codyco-superbuild/build/libraries/iDynTree/lib/libidyntree-high-level-kdl.so->/homes/romeres/Repositories/codyco-superbuild/build/libraries/iDynTree/lib/iDynTreeMEX.mexa64'
Missing symbol '_ZNSt7__cxx1118basic_stringstreamIcSt11char_traitsIcESaIcEED1Ev' required by
'/homes/romeres/Repositories/codyco-superbuild/build/libraries/iDynTree/lib/libidyntree-core.so->/homes/romeres/Repositories/codyco-superbuild/build/libraries/iDynTree/lib/libidyntree-high-level-kdl.so->/homes/romeres/Repositories/codyco-superbuild/build/libraries/iDynTree/lib/iDynTreeMEX.mexa64'
Missing symbol '_ZNSt7__cxx1119basic_ostringstreamIcSt11char_traitsIcESaIcEED1Ev' required by
'/homes/romeres/Repositories/codyco-superbuild/build/libraries/iDynTree/lib/libidyntree-core.so->/homes/romeres/Repositories/codyco-superbuild/build/libraries/iDynTree/lib/libidyntree-high-level-kdl.so->/homes/romeres/Repositories/codyco-superbuild/build/libraries/iDynTree/lib/iDynTreeMEX.mexa64'
Missing symbol '_ZTTNSt7__cxx1118basic_stringstreamIcSt11char_traitsIcESaIcEEE' required by
'/homes/romeres/Repositories/codyco-superbuild/build/libraries/iDynTree/lib/libidyntree-core.so->/homes/romeres/Repositories/codyco-superbuild/build/libraries/iDynTree/lib/libidyntree-high-level-kdl.so->/homes/romeres/Repositories/codyco-superbuild/build/libraries/iDynTree/lib/iDynTreeMEX.mexa64'
Missing symbol '_ZTTNSt7__cxx1119basic_ostringstreamIcSt11char_traitsIcESaIcEEE' required by
[... cut ...]
Error in iCubLeftArmFTRegressor (line 6)
regrGen = iDynTree.DynamicsRegressorGenerator();
```
Initially I encountered the problem with CMake 3.5.1 but also after updating the version the problem remained. I tried to follow a workaround similar to @lucaTagliapietra and I added the following commands:

```
addpath('/homes/romeres/Repositories/codyco-superbuild/build/libraries/iDynTree/lib')
addpath('/homes/romeres/Repositories/codyco-superbuild/libraries/iDynTree/bindings/matlab/autogenerated')
```
but it didn't work.

traversaro, 2018-03-21:
Hi @DiegoRomeres ! 
  I fear that this is another instance of the TROMH (as @diegoferigo called it, i.e. the "The Root Of Many Headache"). You can find a workaround in https://github.com/robotology/codyco-superbuild/issues/141 .

diegoferigo, 2018-03-21:
Other pointer to it https://robotology.github.io/wb-toolbox/mkdocs/troubleshooting/#problems-finding-libraries-and-libstdc