gbionics/idyntree #397: Update InstallBasicPackageFiles.cmake to allow embedded installs

ahoarau, 2017-11-30:
In ORCA https://github.com/syroco/orca i'm embedding idyntree as an external dependencies
(via`add_subdirectories(external/idyntree)` ).

In `InstallBasicPackageFiles.cmake` it'd be great to change all the `${CMAKE_SOURCE_DIR}` to `${PROJECT_SOURCE_DIR}` and `${CMAKE_BINARY_DIR}` to `${PROJECT_BINARY_DIR}`. Now they point to the containing repo (orca) and not the iDynTree project (what we actually want).

Right now, if iDynTree is embedded, the `iDynTreeConfig.cmake.in` is not found at the root of the repo (ref https://github.com/robotology/idyntree/blob/master/cmake/InstallBasicPackageFiles.cmake#L273) and a standard one is generated with some missing values.

The commit in question :
https://github.com/syroco/orca/commit/d7b51bf0b9dffacbd1deeb168c33bda304e9fab2

traversaro, 2017-11-30:
I think this make sense, but we need probably to be added as a option in InstallBasicPackageFiles.cmake at YCM level, because I am bit afraid there could be projects out in the wild using  multiple `project(..)` calls inside the same CMake project. 

Unrelated to this, have you considered using the ExternalProject CMake module to compile the vendored versions of your dependencies? ExternalProject can be configured to use the source directory inside your repository, and is nice because you avoid polluting your CMake project with target and variables from the vendored dependencies.

traversaro, 2017-11-30:
See https://bitbucket.org/ignitionrobotics/ign-cmake/src/15b9092f1a11f15745db6b1e81f3f32edbbdf8a2/examples/CMakeLists.txt?at=default&fileviewer=file-view-default#CMakeLists.txt-30 for an example of such a use of ExternalProject .

ahoarau, 2017-11-30:
Yup I tried that first, but I failed to extract the interesting variables, like `iDynTree::idyntree-high-level` for instance.
+ When you export your project (that uses idyntree), it returns an error saying that idyntree is missing in the export set because was not built in this repo.

I am surely using externalProject the wrong way

ahoarau, 2017-11-30:
> I think this make sense, but we need probably to be added as a option in InstallBasicPackageFiles.cmake at YCM level, because I am bit afraid there could be projects out in the wild using multiple project(..) calls inside the same CMake project.

I'm very much in favor of having an option :+1:

traversaro, 2019-04-05:
This have been fixed at the YCM level ( https://github.com/robotology/ycm/issues/162  ), we just need to update the vendored files in iDynTree.