gbionics/idyntree #400: Cleanup optimalcontrol component

traversaro, 2017-12-06:
See comments to https://github.com/robotology/idyntree/pull/399/files .

traversaro, 2017-12-07:
We should also remember to update the changelog ( https://github.com/robotology/idyntree/blob/devel/doc/releases/v0_10.md ) to mention the new optimal control classes and the modifications to the spline classes.

traversaro, 2018-06-13:
Any idea about this comments @S-Dafarra : 
 * https://github.com/robotology/idyntree/pull/399/files#r155391525 
 * https://github.com/robotology/idyntree/pull/399/files#r155391745 

?

S-Dafarra, 2018-06-13:
Regarding the first, I will delete that comment in a (near) future PR.

Regarding the second, I already changed it https://github.com/robotology/idyntree/blob/devel/src/optimalcontrol/tests/CMakeLists.txt

traversaro, 2019-08-02:
All the comments fixed in `devel` .

closed 2019-08-02 by traversaro