gbionics/idyntree #404: Change KinDynComputations model's base link

diegoferigo, 2017-12-12:
Right now the default base link used by iDynTree is the [first link loaded from the urdf](https://github.com/robotology/idyntree/blob/master/src/model/src/Model.cpp#L255-L258). In most cases this matches to the `root_link`. It is however possible to specify an alternative base link using [`Model::setDefaultBaseLink`](https://github.com/robotology/idyntree/blob/master/src/model/src/Model.cpp#L586).

`KinDynComputations` contains a [`setFloatingBase`](https://github.com/robotology/idyntree/blob/master/src/high-level/src/KinDynComputations.cpp#L423-L427) method that should allow specifying and alternative base link. It doesn't seem this method actually changes it, am I missing something? @traversaro 

cc @gabrielenava

traversaro, 2017-12-12:
> Right now the default base link used by iDynTree is the first link loaded from the urdf.

For the sake of precision, for URDF-loaded model the default base link is always the root link of the URDF tree: https://github.com/robotology/idyntree/blob/a982b71e0adfd7f2f32eb152bde70922fbb24389/src/model_io/urdf/src/URDFModelImport.cpp#L744 . 

> KinDynComputations contains a setFloatingBase method that should allow specifying and alternative base link. It doesn't seem this method actually changes it, am I missing something? 

This line https://github.com/robotology/idyntree/blob/master/src/high-level/src/KinDynComputations.cpp#L426 overwrites `this->pimpl->m_traversal`, that is encoding the base link used for the KinDynComputations.

traversaro, 2017-12-12:
Please close the issue if the reply answer your doubts.

diegoferigo, 2017-12-14:
Clear! Thanks.

closed 2017-12-14 by diegoferigo