gbionics/idyntree #405: The CubicSpline is not considering the specified initial acceleration

S-Dafarra, 2017-12-13:
Even if the methods ``setInitialConditions`` and ``setFinalConditions``ask for the initial and final acceleration, they are not used in the code. Indeed the acceleration is continuous in every via points, but its initial and final value is different from the desired ones (and usually different from zero).

The implementation follows what is written here https://www.dis.uniroma1.it/~deluca/rob1_en/13_TrajectoryPlanningJoints.pdf from slide 20. Apparently there is the possibility to extend the implemented algorithm to take into account also the desired boundary accelerations, but I am still investigating it.

GiulioRomualdi, 2018-04-17:
In section 4.4.4 of [Trajectory Planning for Automatic Machines and Robots](https://www.springer.com/it/book/9783540856283) the authors explain how to extend the standard spline algorithm in order to take into account the desired boundary accelerations. 

The proposed method can be modified and easily implemented on the `iDynTree::CubicSpline` class.

I can take care of this issue.

S-Dafarra, 2018-04-17:
Great! Indeed I was also searching for a nice reference (not a set of slides) for the implemented spline!

GiulioRomualdi, 2019-06-12:
cc @MiladShafiee