gbionics/idyntree #406: iDynTree not compiling in devel on Ubuntu 16.04

traversaro, 2017-12-14:
~~~
[ 50%] Building CXX object src/sensors/CMakeFiles/idyntree-sensors.dir/src/PredictSensorsMeasurements.cpp.o
[ 50%] Building CXX object src/sensors/CMakeFiles/idyntree-sensors.dir/src/ModelSensorsTransformers.cpp.o
<http://10.255.72.58:8080/job/superbuild-isaac/ws/isaac/iDynTree/src/sensors/src/ModelSensorsTransformers.cpp>: In function ‘bool iDynTree::createReducedModelAndSensors(const iDynTree::Model&, const iDynTree::SensorsList&, const std::vector<std::__cxx11::basic_string<char> >&, iDynTree::Model&, iDynTree::SensorsList&)’:
<http://10.255.72.58:8080/job/superbuild-isaac/ws/isaac/iDynTree/src/sensors/src/ModelSensorsTransformers.cpp>:41:10: warning: ‘auto’ changes meaning in C++11; please remove it [-Wc++0x-compat]
    for (auto it = fullSensors.sensorsIteratorForType(SIX_AXIS_FORCE_TORQUE); it.isValid(); ++it) {
         ^
<http://10.255.72.58:8080/job/superbuild-isaac/ws/isaac/iDynTree/src/sensors/src/ModelSensorsTransformers.cpp>:41:15: error: ‘it’ does not name a type
    for (auto it = fullSensors.sensorsIteratorForType(SIX_AXIS_FORCE_TORQUE); it.isValid(); ++it) {
              ^
<http://10.255.72.58:8080/job/superbuild-isaac/ws/isaac/iDynTree/src/sensors/src/ModelSensorsTransformers.cpp>:41:79: error: expected ‘;’ before ‘it’
    for (auto it = fullSensors.sensorsIteratorForType(SIX_AXIS_FORCE_TORQUE); it.isValid(); ++it) {
                                                                              ^
<http://10.255.72.58:8080/job/superbuild-isaac/ws/isaac/iDynTree/src/sensors/src/ModelSensorsTransformers.cpp>:41:79: error: ‘it’ was not declared in this scope
src/sensors/CMakeFiles/idyntree-sensors.dir/build.make:230: recipe for target 'src/sensors/CMakeFiles/idyntree-sensors.dir/src/ModelSensorsTransformers.cpp.o' failed
~~~

Probably due to https://github.com/robotology/idyntree/pull/325 .

traversaro, 2017-12-14:
cc @diegoferigo

diegoferigo, 2017-12-14:
I thought iDynTree supported `C++11` at least in the source files. What about that `superbuild-isaac`?

traversaro, 2017-12-14:
Apparently it was enabled just on some parts. I added a hotfix in https://github.com/robotology/idyntree/commit/139261c6b3494666c451221e24d714923bbe6eb2 , but I think we can now safely enable it everywhere (including the headers).

diegoferigo, 2017-12-15:
Well done @traversaro. For some reason this issue didn't close automatically. Closing.

closed 2017-12-15 by diegoferigo