gbionics/idyntree #423: [Matlab] Missing dependent shared libraries + Missing symbols

aikolina, 2018-03-12:
I tried a simple `iDynTree` command in Matlab, but I get the following error:

```MATLAB
>> linkToAdd = iDynTree.Link();
Invalid MEX-file '/home/adinale/src/robotology-superbuild/build/install/mex/iDynTreeMEX.mexa64': Missing dependent shared libraries:
'libidyntree-core.so' required by
'/home/adinale/src/robotology-superbuild/build/install/lib/libidyntree-high-level-kdl.so->/home/adinale/src/robotology-superbuild/build/install/mex/iDynTreeMEX.mexa64'
'libidyntree-sensors.so' required by
'/home/adinale/src/robotology-superbuild/build/install/lib/libidyntree-regressors.so->/home/adinale/src/robotology-superbuild/build/install/mex/iDynTreeMEX.mexa64'
'libidyntree-model.so' required by
'/home/adinale/src/robotology-superbuild/build/install/lib/libidyntree-visualization.so->/home/adinale/src/robotology-superbuild/build/install/mex/iDynTreeMEX.mexa64'
'libidyntree-core.so' required by
'/home/adinale/src/robotology-superbuild/build/install/lib/libidyntree-visualization.so->/home/adinale/src/robotology-superbuild/build/install/mex/iDynTreeMEX.mexa64'
'libidyntree-modelio-urdf.so' required by
'/home/adinale/src/robotology-superbuild/build/install/lib/libidyntree-high-level.so->/home/adinale/src/robotology-superbuild/build/install/mex/iDynTreeMEX.mexa64'
'libidyntree-model.so' required by
'/home/adinale/src/robotology-superbuild/build/install/lib/libidyntree-high-level.so->/home/adinale/src/robotology-superbuild/build/install/mex/iDynTreeMEX.mexa64'
'libidyntree-core.so' required by
'/home/adinale/src/robotology-superbuild/build/install/lib/libidyntree-high-level.so->/home/adinale/src/robotology-superbuild/build/install/mex/iDynTreeMEX.mexa64'
'libidyntree-modelio-urdf.so' required by
'/home/adinale/src/robotology-superbuild/build/install/lib/libidyntree-estimation.so->/home/adinale/src/robotology-superbuild/build/install/mex/iDynTreeMEX.mexa64'
'libidyntree-sensors.so' required by
'/home/adinale/src/robotology-superbuild/build/install/lib/libidyntree-estimation.so->/home/adinale/src/robotology-superbuild/build/install/mex/iDynTreeMEX.mexa64'
'libidyntree-model.so' required by
'/home/adinale/src/robotology-superbuild/build/install/lib/libidyntree-estimation.so->/home/adinale/src/robotology-superbuild/build/install/mex/iDynTreeMEX.mexa64'
'libidyntree-core.so' required by
'/home/adinale/src/robotology-superbuild/build/install/lib/libidyntree-estimation.so->/home/adinale/src/robotology-superbuild/build/install/mex/iDynTreeMEX.mexa64'
'libidyntree-sensors.so' required by
'/home/adinale/src/robotology-superbuild/build/install/lib/libidyntree-modelio-urdf.so->/home/adinale/src/robotology-superbuild/build/install/mex/iDynTreeMEX.mexa64'
'libidyntree-model.so' required by
'/home/adinale/src/robotology-superbuild/build/install/lib/libidyntree-modelio-urdf.so->/home/adinale/src/robotology-superbuild/build/install/mex/iDynTreeMEX.mexa64'
'libidyntree-core.so' required by
'/home/adinale/src/robotology-superbuild/build/install/lib/libidyntree-modelio-urdf.so->/home/adinale/src/robotology-superbuild/build/install/mex/iDynTreeMEX.mexa64'
'libidyntree-model.so' required by
'/home/adinale/src/robotology-superbuild/build/install/lib/libidyntree-sensors.so->/home/adinale/src/robotology-superbuild/build/install/mex/iDynTreeMEX.mexa64'
'libidyntree-core.so' required by
'/home/adinale/src/robotology-superbuild/build/install/lib/libidyntree-sensors.so->/home/adinale/src/robotology-superbuild/build/install/mex/iDynTreeMEX.mexa64'
'libidyntree-core.so' required by
'/home/adinale/src/robotology-superbuild/build/install/lib/libidyntree-model.so->/home/adinale/src/robotology-superbuild/build/install/mex/iDynTreeMEX.mexa64'

Missing symbol '_ZNSt12out_of_rangeC1EPKc' required by '/home/adinale/src/robotology-superbuild/build/install/mex/iDynTreeMEX.mexa64'
Missing symbol '_ZNSt16invalid_argumentC1EPKc' required by '/home/adinale/src/robotology-superbuild/build/install/mex/iDynTreeMEX.mexa64'
Missing symbol '_ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE4swapERS4_' required by
'/home/adinale/src/robotology-superbuild/build/install/mex/iDynTreeMEX.mexa64'
Missing symbol '_ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE9_M_appendEPKcm' required by
'/home/adinale/src/robotology-superbuild/build/install/mex/iDynTreeMEX.mexa64'
Missing symbol '_ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE9_M_assignERKS4_' required by
'/home/adinale/src/robotology-superbuild/build/install/mex/iDynTreeMEX.mexa64'
Missing symbol '_ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE9_M_createERmm' required by
'/home/adinale/src/robotology-superbuild/build/install/mex/iDynTreeMEX.mexa64'
Missing symbol '_ZNSt7__cxx1118basic_stringstreamIcSt11char_traitsIcESaIcEED1Ev' required by
'/home/adinale/src/robotology-superbuild/build/install/mex/iDynTreeMEX.mexa64'
Missing symbol '_ZTTNSt7__cxx1118basic_stringstreamIcSt11char_traitsIcESaIcEEE' required by
'/home/adinale/src/robotology-superbuild/build/install/mex/iDynTreeMEX.mexa64'
Missing symbol '_ZTVNSt7__cxx1115basic_stringbufIcSt11char_traitsIcESaIcEEE' required by
'/home/adinale/src/robotology-superbuild/build/install/mex/iDynTreeMEX.mexa64'
Missing symbol '_ZTVNSt7__cxx1118basic_stringstreamIcSt11char_traitsIcESaIcEEE' required by
'/home/adinale/src/robotology-superbuild/build/install/mex/iDynTreeMEX.mexa64'
Missing symbol '_ZN8iDynTree19LinearMotionVector3C1Ev' required by
'/home/adinale/src/robotology-superbuild/build/install/lib/libidyntree-high-level-kdl.so->/home/adinale/src/robotology-superbuild/build/install/mex/iDynTreeMEX.mexa64'
Missing symbol '_ZN8iDynTree20RotationalInertiaRawC1Ev' required by
'/home/adinale/src/robotology-superbuild/build/install/lib/libidyntree-high-level-kdl.so->/home/adinale/src/robotology-superbuild/build/install/mex/iDynTreeMEX.mexa64'
Missing symbol '_ZN8iDynTree29AngularMotionVector3SemanticsC1Ev' required by
'/home/adinale/src/robotology-superbuild/build/install/lib/libidyntree-high-level-kdl.so->/home/adinale/src/robotology-superbuild/build/install/mex/iDynTreeMEX.mexa64'
Missing symbol '_ZN8iDynTree5TwistC1ERKNS_19SpatialMotionVectorE' required by
'/home/adinale/src/robotology-superbuild/build/install/lib/libidyntree-high-level-kdl.so->/home/adinale/src/robotology-superbuild/build/install/mex/iDynTreeMEX.mexa64'
Missing symbol '_ZN8iDynTree5TwistC1Ev' required by
'/home/adinale/src/robotology-superbuild/build/install/lib/libidyntree-high-level-kdl.so->/home/adinale/src/robotology-superbuild/build/install/mex/iDynTreeMEX.mexa64'
Missing symbol '_ZN8iDynTree8Position4ZeroEv' required by
'/home/adinale/src/robotology-superbuild/build/install/lib/libidyntree-high-level-kdl.so->/home/adinale/src/robotology-superbuild/build/install/mex/iDynTreeMEX.mexa64'
[... 1557 lines cut ...]
Missing symbol '_ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE4swapERS4_' required by
'/home/adinale/src/robotology-superbuild/build/install/lib/libidyntree-core.so->/home/adinale/src/robotology-superbuild/build/install/mex/iDynTreeMEX.mexa64'
Missing symbol '_ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE9_M_assignERKS4_' required by
'/home/adinale/src/robotology-superbuild/build/install/lib/libidyntree-core.so->/home/adinale/src/robotology-superbuild/build/install/mex/iDynTreeMEX.mexa64'
Missing symbol '_ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE9_M_createERmm' required by
'/home/adinale/src/robotology-superbuild/build/install/lib/libidyntree-core.so->/home/adinale/src/robotology-superbuild/build/install/mex/iDynTreeMEX.mexa64'
Missing symbol '_ZNSt7__cxx1118basic_stringstreamIcSt11char_traitsIcESaIcEED1Ev' required by
'/home/adinale/src/robotology-superbuild/build/install/lib/libidyntree-core.so->/home/adinale/src/robotology-superbuild/build/install/mex/iDynTreeMEX.mexa64'
Missing symbol '_ZNSt7__cxx1119basic_ostringstreamIcSt11char_traitsIcESaIcEED1Ev' required by
'/home/adinale/src/robotology-superbuild/build/install/lib/libidyntree-core.so->/home/adinale/src/robotology-superbuild/build/install/mex/iDynTreeMEX.mexa64'
Missing symbol '_ZTTNSt7__cxx1118basic_stringstreamIcSt11char_traitsIcESaIcEEE' required by
'/home/adinale/src/robotology-superbuild/build/install/lib/libidyntree-core.so->/home/adinale/src/robotology-superbuild/build/install/mex/iDynTreeMEX.mexa64'
Missing symbol '_ZTTNSt7__cxx1119basic_ostringstreamIcSt11char_traitsIcESaIcEEE' required by
[... cut ...]
'/home/adinale/src/robotology-superbuild/build/install/lib/libidyntree-core.so->/home/adinale/src/robotology-superbuild/build/install/mex/iDynTreeMEX.mexa64'
Missing symbol '_ZTVNSt7__cxx1119basic_ostringstreamIcSt11char_traitsIcESaIcEEE' required by
'/home/adinale/src/robotology-superbuild/build/install/lib/libidyntree-core.so->/home/adinale/src/robotology-superbuild/build/install/mex/iDynTreeMEX.mexa64'.

Error in iDynTree.Link (line 12)
        tmp = iDynTreeMEX(870, varargin{:});
```
[:exclamation:] I have installed `iDynTree` using the `robotology-superbuild`.

traversaro, 2018-03-12:
See: 
  * https://robotology.github.io/wb-toolbox/mkdocs/troubleshooting/#problems-finding-libraries-and-libstdc 
 * https://github.com/robotology/codyco-superbuild/issues/141   
* https://github.com/robotology/wb-toolbox/issues/101

aikolina, 2018-03-12:
I solved the problem after discussing with @diegoferigo.

I used the solution suggested in [this issue](https://github.com/robotology/wb-toolbox/issues/101#issuecomment-372016880)!

@traversaro Always super efficient! Thanks! :smile:

closed 2018-03-12 by traversaro

diegoferigo, 2018-03-12:
Would it be worth to include this information also in the [idyntree matlab bindings section](https://github.com/robotology/idyntree#matlab)?

Unfortunately this is the case where the Matlab error does not tell you directly the problem (missing `libstdc++`).