gbionics/idyntree #424: Use Joint constructor in Matlab/Octave

traversaro, 2018-03-13:
Originally posted by @aikolina  in https://github.com/robotology/idyntree/issues/175 :+1: 

I have a problem to use the RevoluteJoint constructor in Matlab.

I have tried [all the possible variants](http://wiki.icub.org/codyco/dox/html/idyntree/html/classiDynTree_1_1RevoluteJoint.html#aa09aee905d118d03a9d22b5bdc98e132) (two/four/no arguments), and I always get the error message **No matching constructor**.

```Matlab
>> planar_model
Error using iDynTree.RevoluteJoint (line 85)
No matching constructor

Error in planar_model (line 37)
ankleJoint = iDynTree.RevoluteJoint();
 
>> planar_model
Error using iDynTree.RevoluteJoint (line 85)
No matching constructor

Error in planar_model (line 38)
ankleJoint = iDynTree.RevoluteJoint(baseLink, lowerLegLink, baseLink_H_lowerLegLink, rotationAxis);
 
>> planar_model
Error using iDynTree.RevoluteJoint (line 85)
No matching constructor

Error in planar_model (line 39)
ankleJoint = iDynTree.RevoluteJoint(baseLink_H_lowerLegLink, rotationAxis);

```

traversaro, 2018-03-13:
I reproduced the problem, for some reason SWIG is not generating the constructor for this class. Working on it.

traversaro, 2018-03-13:
The problem is affecting also the Python bindings:
~~~
44: ======================================================================
44: ERROR: testPrismaticJoint (__main__.JointTest)
44: ----------------------------------------------------------------------
44: Traceback (most recent call last):
44:   File "joints.py", line 22, in testPrismaticJoint
44:     prismJoint = iDynTree.PrismaticJoint();
44: AttributeError: 'module' object has no attribute 'PrismaticJoint'
44: 
44: ======================================================================
44: ERROR: testRevoluteJoint (__main__.JointTest)
44: ----------------------------------------------------------------------
44: Traceback (most recent call last):
44:   File "joints.py", line 25, in testRevoluteJoint
44:     revJoint = iDynTree.RevoluteJoint();
44:   File "../../python/iDynTree.py", line 4220, in __init__
44:     raise AttributeError("No constructor defined - class is abstract")
44: AttributeError: No constructor defined - class is abstract
44: 
44: ----------------------------------------------------------------------
~~~

This error is a bit more clear. Apparently, for some reason SWIG thinks that the class is abstract and does not generate a constructor.

traversaro, 2018-03-13:
Related: https://mail.python.org/pipermail/python-list/2008-April/476273.html .

traversaro, 2018-03-13:
Apparently the problem is also affecting the `FixedJoint` class, and in that case the inheritance structure is easier to debug.

traversaro, 2018-03-13:
Apparently there were some swig warnings I did not noticed:
~~~
/home/straversaro/src/robotology-superbuild/robotology/iDynTree/src/model/include/iDynTree/Model/FixedJoint.h:27: Warning 403: Class 'iDynTree::FixedJoint' might be abstract, no constructors generated,
/home/straversaro/src/robotology-superbuild/robotology/iDynTree/src/model/include/iDynTree/Model/IJoint.h:339: Warning 403: Method iDynTree::IJoint::enablePosLimits(bool) might not be implemented.
/home/straversaro/src/robotology-superbuild/robotology/iDynTree/src/model/include/iDynTree/Model/RevoluteJoint.h:30: Warning 403: Class 'iDynTree::RevoluteJoint' might be abstract, no constructors generated,
/home/straversaro/src/robotology-superbuild/robotology/iDynTree/src/model/include/iDynTree/Model/IJoint.h:339: Warning 403: Method iDynTree::IJoint::enablePosLimits(bool) might not be implemented.
~~~

aikolina, 2018-03-13:
Actually I noticed the same misbehaviour with Prismatic Joint too. -_-"

traversaro, 2018-03-13:
PrismaticJoint is even worse, as it is not wrapped at all by Swig.

traversaro, 2018-03-13:
Fixed in https://github.com/robotology/idyntree/pull/425 , please add my fork as a remote to check the branch @aikolina !

traversaro, 2018-04-07:
Fixed by https://github.com/robotology/idyntree/pull/425 .

closed 2018-04-07 by traversaro