gbionics/idyntree #426: Strange behavior of YARPConversions templated functions

diegoferigo, 2018-03-16:
I'm facing a strange behavior in the `YARPConversion` helper function.

`iDynTree` has generic templates, such as: 

https://github.com/robotology/idyntree/blob/d4a1cf98a80391937da6a040982c72764aec9a86/src/yarp/include/iDynTree/yarp/YARPConversions.h#L178-L185

and functions, which their presence should prevent the template to be instantiated. An example matching the template taken as an example is:

https://github.com/robotology/idyntree/blob/d4a1cf98a80391937da6a040982c72764aec9a86/src/yarp/include/iDynTree/yarp/YARPConversions.h#L113-L122

https://github.com/robotology/idyntree/blob/d4a1cf98a80391937da6a040982c72764aec9a86/src/yarp/src/YARPConversions.cpp#L109-L114

However, debugging some code which uses this function, I realized the regular function is never called. Instead, for some reason, the template is instantiated. This caused a problem because the difference between the two implementations is that the template fails when the two container do not have the same size (I was just reserving, not resizing the `iDynTree::VecDynSize` object).

Probably in this case, using template specialization would work smoother, but it is just a guess. I still don't get why the instantiation happens in the first place. Any idea? @traversaro

diegoferigo, 2018-03-16:
Writing down the problem often helps solving by yourself the issue. After submitting this, I had the idea to check what function get instantiated, with this outcome:

```cpp
bool iDynTree::toiDynTree<iDynTree::JointPosDoubleArray>(yarp::sig::Vector const&, iDynTree::JointPosDoubleArray&)
```

Despite ` class JointPosDoubleArray: public VectorDynSize`, the template seems to have an higher priority regardless the inheritance.

If no one else has other comments, feel free to close this issue.

traversaro, 2019-04-25:
This is indeed tricky, I did not remember reading this issue at the time it was opened. Ideally `toiDynTree` should work in the same way for `VectorDynSize` or any  class that inherits publicly  from it. Do you  mind if we keep this issue open?

diegoferigo, 2019-04-25:
> Ideally toiDynTree should work in the same way for VectorDynSize or any class that inherits publicly from it

Definitely :+1: on this

> Do you mind if we keep this issue open?

You're affecting my spring cleaning process :laughing: Not at all :)