gbionics/idyntree #427: Migration to native classes: Use IDYNTREE_USES_KDL Cmake Option Where?

mouhyemen, 2018-03-28:
[WARNING] The iDynTree::Regressors::DynamicsRegressorGenerator class still needs to be migrated to
[WARNING] iDynTree native classes. To continue to use the iDynTree::HighLevel::DynamicsComputations class in the 
[WARNING] bindings in the current form, please compile iDynTree with the IDYNTREE_USES_KDL CMake option.
[WARNING] The iDynTree::Regressors::DynamicsRegressorGenerator class still needs to be migrated to
[WARNING] iDynTree native classes. To continue to use the iDynTree::HighLevel::DynamicsComputations class in the 
[WARNING] bindings in the current form, please compile iDynTree with the IDYNTREE_USES_KDL CMake option.

I keep getting the above warnings and i don't think iDynTree scripts are running effectively. I have compiled and installed iDynTree from CoDyCo superbuild as well as iDynTree superbuild.

In both the cases, I did not find any IDYNTREE_USES_KDL Cmake option? What's going on?

traversaro, 2018-03-29:
Hi @mouhyemen , `IDYNTREE_USES_KDL` is currently a deprecated CMake option, and as such is not exposed from the superbuild. If you want to enable it, you need to manually go in the idyntree subproject build directory (i.e. <superbuild_dir>/build/libraries/idyntree`), run `ccmake ..` there and enable it. 
However, I noticed that FloBaRoID still refers to the `idyntree-superbuild` as the preferred way to install iDynTree. For this reason, I exposed the `IDYNTREE_USES_KDL` option in idyntree-superbuild in https://github.com/robotology/idyntree-superbuild/commit/a61ba526fab6dcb01ac24cedfb9d070fc59982c2 . Note that we will need orocos_kdl and urdfdom installed on your computer, if you are using Ubuntu Linux 16.04 you can obtain them installing the `liborocos-kdl-dev` and `liburdfdom-dev` packages.  

The regressor part of iDynTree will be migrated not to use KDL (hopefully) during the 2018. 

Let me know if this solved your problem.

mouhyemen, 2018-03-29:
Hey,
I am getting the following CMake error while configuring. I cloned the latest repo because I was getting some other error by going into **iDynTree subproject** and running ccmake with the existing CMakeLists.txt.
[image]

traversaro, 2018-03-29:
That is a warning, we are going to solve it, but for the time being you can ignore it by pressing `e`.

mouhyemen, 2018-03-29:
Thanks! I got it installed using IDYNTREE_USES_KDL option.
I also installed the **liborocos-kdl-dev** and **liburdfdom-dev** packages.

traversaro, 2018-07-04:
Closing as it seems that all the problems have been solved @mouhyemen feel free to reopen or open a new issue if you think it will be necessary.

closed 2018-07-04 by traversaro