gbionics/idyntree #429: Calculating the Inertia for a composite rigid body

vibhoraggarwal, 2018-03-29:
It seems that there does not exist a function, which could help me calculate the Inertia of the composite rigid bodies that are connected to the link.

For example, Calculating the Inertia for the bodies connected to the `r_ankle_pitch` joint, then I need to calculate combined Inertia for all the bodies(links) connected to it(`r_ankle_1, r_ankle_2, r_foot ..`)  

cc: @gabrielenava @traversaro

gabrielenava, 2018-03-29:
You can have a look the Matlab script that we used long time ago to check if the inertia of the [new icub model]() was the same of that one of the [old model]().

I attached the script to the issue.

[checkModelFrames.m.zip](https://github.com/robotology/idyntree/files/1861067/checkModelFrames.m.zip)

vibhoraggarwal, 2018-04-05:
The file `checkModelFrames.m.zip` mentioned above is just for checking any error in calculated Inertia.

For calculating the Inertia for the composite rigid body, I would also need to access the `parent array` for the robot, which I am not able to access. Could you refer me to some documentation which can guide me to access the parent array?
@gabrielenava @traversaro

vibhoraggarwal, 2018-04-20:
I was able to calculate the `Inertia` of the composite bodies in iCub. The files for its calculation could be found, [clicking here](https://github.com/loc2/element_joint-torque-control/tree/master/Computing_mass_matrixandBiasforces)

gabrielenava, 2018-06-19:
Hi @vibhoraggarwal, do you think we can close this issue?

vibhoraggarwal, 2018-06-19:
@gabrielenava Yes, the task was successful; I have also updated the link in the [above comment](https://github.com/robotology/idyntree/issues/429#issuecomment-383062136). Hence, closing the issue. 
Thanks

closed 2018-06-19 by vibhoraggarwal