gbionics/idyntree #430: Implement externally owned vector and matrices

traversaro, 2018-03-30:
To support reading and writing from and to Vector and Matrices already allocated in a form different from `iDynTree::VectorDynSize`, `iDynTree::MatrixDynSize`, `iDynTree::VectorFixSize`, `iDynTree::MatrixFixSize` (such as `Eigen::MatrixXd`, `std::vector<double>` , `std::array<double>`, raw c-buffers, numpy matrices, etc etc, it may be useful to define classes to represent Vector and Matrices externally owned. 

While the matrix implementation is probably more complicated, the Vector implementation could be probably as simple as: 
~~~
namespace iDynTree
{
class VectorExt
{
private:
    double * m_data;
    size_t m_size;
public:
    VectorExt(double * data, size_t size); 
}
}
~~~

The most used methods of the classes such as KinDynComputations and InverseKinematics can be modified to accept this VectorExt class, to simplify interoperability and reducing the need of unnecessary copy, while avoiding dealing directly with raw pointers and preserving the possibility of checking sizes. 

A thing that needs  to be check is how to properly differentiate external buffers that can be modified to external buffers that can be only read.

traversaro, 2018-03-30:
Related discussions: 
* https://github.com/robotology/yarp/pull/1582#pullrequestreview-102847779
* https://github.com/robotology/yarp/issues/76
* https://github.com/robotology/codyco-modules/issues/26#issuecomment-43882203 
* https://abseil.io/tips/1 (`std::string_view` is basically the same concept, but applied to strings)

traversaro, 2018-03-30:
Related discussion at C++ standard committee : 
* `array_view` proposal (generalization of what proposed here to multidimensional arrays, including aspect of stride, now abandoned due to its complexity): http://www.open-std.org/jtc1/sc22/wg21/docs/papers/2014/n3851.pdf
* `span` proposal: similar to `array_view`, but only for 1D sequences: http://www.open-std.org/jtc1/sc22/wg21/docs/papers/2014/n3851.pdf

S-Dafarra, 2018-04-09:
Regarding the vector, it seems to me that ``VectorDynSize`` has already the structure of  ``VectorExt``. Do you think it may be a possibility to implement "maps" as methods of ``VectorDynSize``?

In this case 
> A thing that needs to be check is how to properly differentiate external buffers that can be modified to external buffers that can be only read.

would still be an open issue.

traversaro, 2018-04-09:
> Regarding the vector, it seems to me that VectorDynSize has already the structure of VectorExt. Do you think it may be a possibility to implement "maps" as methods of VectorDynSize?

VectorDynSize supports the `resize` method, and so it cannot be adapted to be a "VectorExt" without changing drastically its meaning and probably breaking a lot of code that use it. 

> A possibility could be to use a private flag to discern whether the vector can be resized or not (using asserts similarly to what Eigen does for the maps).

I check the proposal, and the `std::span` proposal is quite reasonable and address also these points , and will be probably accepted in C++20. A possible strategy is to just copy a liberally licensed existing implementation of span or implement and class with the same interface. 

On the other hand, `mdspan` (the equivalent of span for n-order tensors, for n greater than one) is quite complicated and it is not clear if it will be eventually integrated in the standard, so I am afraid that rolling our solution for it may be the best option.

S-Dafarra, 2018-04-09:
Regarding ``span`` these are two available implementations:
- https://github.com/abseil/abseil-cpp/blob/master/absl/types/span.h in [abseil](https://github.com/abseil/abseil-cpp) as you suggested 
- https://github.com/martinmoene/gsl-lite/blob/master/include/gsl/gsl-lite.hpp#L1212 in [gsl-lite](https://github.com/martinmoene/gsl-lite)

traversaro, 2018-04-09:
The abseil one is Apache 2-licensed, so it is not directly incorporable in the iDynTree codebase (see https://www.dwheeler.com/essays/floss-license-slide.html). Furthermore, I have the impression that extracting the span implementation from the complete class would be quite complicated. 

The gsl-lite one is MIT-licensed, so we can incorporate it directly in the iDynTree. I think we can make iDynTree C++14 only at this point (we discussed about this at a YARP level, so it should not be a problem). The `gsl::span` seems quite indipendent, cleaning up the gsl specific stuff (`gsl_api`-> ` `, `gsl_constexpr14` --> `constexpr, etc etc) I think it could be include in iDynTree as `iDynTree::Span` (I think we can copy the capitalization from abseil to minimize the risk of collisions in code that use both `using namespace iDynTree` and `using namespace std`.

S-Dafarra, 2018-04-09:
> I think it could be include in iDynTree as iDynTree::Span (I think we can copy the capitalization from abseil to minimize the risk of collisions in code that use both using namespace iDynTree and using namespace std

:+1: 
> The gsl::span seems quite indipendent, cleaning up the gsl specific stuff (gsl_api-> , gsl_constexpr14 --> constexpr, etc etc)

Were you thinking about a rough copy/paste or wrapping it as ``iDynTree::Span``?

traversaro, 2018-04-09:
Copy/paste from gsl + cleanup of gsl stuff + renaming as `iDynTree::Span`. 
After that, we can start using it in the interfaces as `iDynTree::Span`.

traversaro, 2018-04-09:
Microsoft's GSL also has a MIT-license implementation of span : https://github.com/Microsoft/GSL/blob/master/include/gsl/span .

traversaro, 2018-04-09:
cc @diegoferigo I suspect this could be useful for wb-toolbox .

S-Dafarra, 2018-04-09:
> Microsoft's GSL also has a MIT-license implementation of span : 

I was looking at [this table](https://github.com/martinmoene/gsl-lite#features) and I did not realize what M stand for :sweat_smile:

traversaro, 2018-04-09:
Yes, because it is already C++14. 

I think we can drop the methods returning using the `std::byte`, substitute `Expects` with `std::assert` and do a general cleanup and probably that one is already read to be a good `iDynTree::Span` .

traversaro, 2018-04-09:
I think we can also drop a few `constexpr` if that is a problem with Visual Studio 2015.

S-Dafarra, 2018-04-09:
``narrow_cast`` can be changed easily into ``static_cast`` (see https://github.com/Microsoft/GSL/blob/master/include/gsl/gsl_util#L92), while  ``narrow`` is a bit more tricky since it checks that the value has not changed after the cast (see https://github.com/Microsoft/GSL/blob/master/include/gsl/gsl_util#L110). Does it make sense to keep this check?

traversaro, 2018-04-09:
I think we can just switch to `static_cast`.

traversaro, 2018-04-09:
But the actual `narrow` call is quite small, so we can copy also that if you feel strongly about it.

S-Dafarra, 2018-04-10:
Considering that in iDynTree we have only vector and matrices as containers, I wonder whether it is better to avoid considering a generic ``multi-span`` class while rather using something that eases the passage from iDynTree to Eigen and eventually``matio`` :thinking: . In this sense, we could use something similar to ``span`` with some additional field to discern, for example,  whether it is row or column major.

traversaro, 2018-04-10:
I totally agree, especially because the ndspan proposal seems to be quite complicated and its inclusion in the standard is uncertain.

S-Dafarra, 2018-04-10:
I would call this new class something like ``matrix_view`` or similar, following the ``string_view`` style. Regarding the implementation, I don't know whether there is some implementation we could already use. If we are going to implement it from scratch, I think I will not be able to have the same performances of ``span`` :sweat_smile:

traversaro, 2018-04-10:
> I think I will not be able to have the same performances of span

I don't know, I don't think we can get too wrong passing a pointer and two integers around. : )
For the `_view` name I don't know, I read somewhere in all the standard proposal that `_view` was associated with the concept of `read-only reference`, while `_ref` was associated with the concept of a reference that could also have write support.  Indeed, the original name for `ndspan` was `arrray_ref` .

traversaro, 2018-04-10:
`Map` is also quite an interesting suffix, given that it is used in Eigen so a lot of developers would be familiar with the concept.

S-Dafarra, 2018-04-10:
I was referring to all the machinery that makes ``span`` and ``make_span`` working effectively with a "general" vector. 

Regarding the name I like the suffix ``Map``. What about ``MatrixMap``?

S-Dafarra, 2018-04-11:
> Assuming that we agree that we don't think to handle the stride stuff at this point

:+1: 

> The compile-time route is attractive for compatibility with Eigen, but it may be problematic for declaring non-header only methods that can take in input both variants.

Sincerely I prefer the compile-time option, also because I don't know if it makes sense for this information to change at run-time. Looking at the ``span`` code I saw line https://github.com/Microsoft/GSL/blob/master/include/gsl/span#L333. If I have undestood correctly (and I have big doubts about that) the public member ``value_type`` gives you information about the type of value stored, even if this information comes from the template. I was thinking of doing the same, i.e. accept the type ``RowMajor`` or ``ColumnMajor`` from template, while exposing a public member referring to it. Do you think this may solve the issue you pointed out?

prashanthr05, 2019-01-03:
An example for iDynTree::Span in action for the use-case of getting and setting states,
``` cpp
bool getState(iDynTree::Span<double>& x) const
{
    x(0) = 0.0;
    return true;
}

bool setState(iDynTree::Span<double>& stateBuf)
{
    // ... check if stateBuf and internal state have same size, else throw error
    // avoiding all that here
   double x = stateBuffer(0);
   return true;
}
```
can be called using, 
```cpp
iDynTree::VectorDynSize state;
state.resize(1);
state(0) = 10.0;
std::cout << state.toString() << std::endl; // prints 10.0
iDynTree::Span<double> buf(state.data(), state.size());
getState(buf);
std::cout << state.toString() << std::endl; // prints 0.0 

state(0) = 10.0;
iDynTree::Span<double> setBuf(state.data(), state.size());
setState(setBuf);
```

traversaro, 2019-01-03:
Exactly! I would however add `asSpan` methods to `VectorDynSize` and `VectorFixSize` .

S-Dafarra, 2019-01-03:
Actually the ``make_span`` method works already for iDynTree vectors. You could do
```cpp
setState(iDynTree::make_span(state))
```

traversaro, 2019-01-03:
You are right, I totally forgot about it!

traversaro, 2019-08-02:
A self-declared "production-ready" implementation of mdspan is available at https://github.com/kokkos/mdspan .

GiulioRomualdi, 2021-12-16:
I think we can close this issue right? 

@traversaro

traversaro, 2021-12-16:
Fixed by:
* https://github.com/robotology/idyntree/pull/734
* https://github.com/robotology/idyntree/pull/434

Thanks @S-Dafarra  @GiulioRomualdi !

closed 2021-12-16 by traversaro