gbionics/idyntree #437: How may I compute the term Jdot in another point ?

ZzhengP, 2018-04-18:
Hello,

I am able to compute a vector that I call "Jdotqdot" in an another point using Jdotqdot.refPoint(vector) in KDL library. I would like to know if something similar exist in IdynTree library.

Regards,

Pu

traversaro, 2018-04-19:
I think the method you are looking for is the [`KinDynComputations::getFrameBiasAcc`](http://robotology.gitlab.io/docs/idyntree/master/classiDynTree_1_1KinDynComputations.html#a81f83749d1b0d4a7c68cfea82ef0491c) method. Note that the semantics of that call depends on the specifiy representation used for FrameVelocity, see http://robotology.gitlab.io/docs/idyntree/master/namespaceiDynTree.html#a0089c6dac34bc6a4f623d9e9e565375a .

At the moment, it cannot accept a new frame on the fly, but you can ad an **additional frame** to the model before specifying the model in the KinDynComputations class, and then call `getFrameBiasAcc` for that additional frame.

ZzhengP, 2018-04-19:
Thank you for your response.

Regards,
Pu

traversaro, 2018-04-19:
Let me know if that solved your problem.

ZzhengP, 2018-04-19:
Before i was trying to create a function like changeRefPoint() in KDL to compute the Jacobian in another point, its work well. But i am not certain that the same function can applied to compute Jdotqdot. I will talk with my colleague for you proposed method. Thank you.

traversaro, 2018-05-20:
Closing for now @ZzhengP , let me know if you have any other question.

closed 2018-05-20 by traversaro