gbionics/idyntree #441: Problems with iDynTree python binding

anqingd, 2018-05-17:
After installing python binding from robotology-superbuild, I tried ```import iDynTree``` in python, but I got errors:
 ```
>>> import iDynTree
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/aduan/robotology-superbuild/build/install/lib/python2.7/dist-packages/iDynTree.py", line 28, in <module>
    _iDynTree = swig_import_helper()
  File "/home/aduan/robotology-superbuild/build/install/lib/python2.7/dist-packages/iDynTree.py", line 24, in swig_import_helper
    _mod = imp.load_module('_iDynTree', fp, pathname, description)
ImportError: libidyntree-high-level-kdl.so: cannot open shared object file: No such file or directory
>>> 

```

traversaro, 2018-05-17:
Thanks @anqingd . I am trying to reproduce the issue.

traversaro, 2018-05-18:
I tried on my setup, and everything seems to work fine. 
Can you please paste: 
* The output of the `ldd <location_where_you_saved_the_superbuild>/robotology-superbuild/build/install/lib/python2.7/dist-packages/_iDynTree.so` ? 
* The output of the `env` command?

traversaro, 2018-05-18:
For reference, this are the output on my machine, where the bindings are working:
~~~bash
straversaro@iiticublap103:~/src/robotology-superbuild/build/install/lib/python2.7/dist-packages$ ldd _iDynTree.so
	linux-vdso.so.1 =>  (0x00007fffbb9c4000)
	libidyntree-high-level-kdl.so => /home/straversaro/src/robotology-superbuild/build/install/lib/libidyntree-high-level-kdl.so (0x00007f7367509000)
	libidyntree-regressors.so => /home/straversaro/src/robotology-superbuild/build/install/lib/libidyntree-regressors.so (0x00007f7367302000)
	libidyntree-visualization.so => /home/straversaro/src/robotology-superbuild/build/install/lib/libidyntree-visualization.so (0x00007f73670fa000)
	libpython2.7.so.1.0 => /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0 (0x00007f7366b6c000)
	libidyntree-high-level.so => /home/straversaro/src/robotology-superbuild/build/install/lib/libidyntree-high-level.so (0x00007f7366953000)
	libidyntree-estimation.so => /home/straversaro/src/robotology-superbuild/build/install/lib/libidyntree-estimation.so (0x00007f73666fa000)
	libidyntree-modelio-urdf.so => /home/straversaro/src/robotology-superbuild/build/install/lib/libidyntree-modelio-urdf.so (0x00007f73664da000)
	libidyntree-sensors.so => /home/straversaro/src/robotology-superbuild/build/install/lib/libidyntree-sensors.so (0x00007f73662b8000)
	libidyntree-model.so => /home/straversaro/src/robotology-superbuild/build/install/lib/libidyntree-model.so (0x00007f7366049000)
	libidyntree-core.so => /home/straversaro/src/robotology-superbuild/build/install/lib/libidyntree-core.so (0x00007f7365dca000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f7365a48000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f7365832000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f7365468000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f736524b000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f7365031000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f7364e2d000)
	libutil.so.1 => /lib/x86_64-linux-gnu/libutil.so.1 (0x00007f7364c2a000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f7364921000)
	libtinyxml.so.2.6.2 => /usr/lib/x86_64-linux-gnu/libtinyxml.so.2.6.2 (0x00007f736470b000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f7367b75000
~~~

~~~bash 
XDG_VTNR=7
LC_PAPER=it_IT.UTF-8
LC_ADDRESS=it_IT.UTF-8
XDG_SESSION_ID=c2
ROBOTOLOGY_SUPERBUILD_ROOT=/home/straversaro/src/robotology-superbuild
XDG_GREETER_DATA_DIR=/var/lib/lightdm-data/straversaro
LC_MONETARY=it_IT.UTF-8
CLUTTER_IM_MODULE=
QT_STYLE_OVERRIDE=gtk
SESSION=xubuntu
GPG_AGENT_INFO=/home/straversaro/.gnupg/S.gpg-agent:0:1
GLADE_PIXMAP_PATH=:
XDG_MENU_PREFIX=xfce-
SHELL=/bin/bash
TERM=xterm
QT_LINUX_ACCESSIBILITY_ALWAYS_ON=1
GAZEBO_MODEL_PATH=/home/straversaro/src/gazebo-superbuild/build/install/share/gazebo-9/models::/home/straversaro/src/robotology-superbuild/build/install/share/gazebo/models
WINDOWID=81788932
LC_NUMERIC=it_IT.UTF-8
UPSTART_SESSION=unix:abstract=/com/ubuntu/upstart-session/1001/2369
GNOME_KEYRING_CONTROL=
GTK_MODULES=gail:atk-bridge
USER=straversaro
LD_LIBRARY_PATH=:/home/straversaro/src/gazebo-superbuild/build/install/lib:/home/straversaro/src/gazebo-superbuild/build/install/lib/gazebo-9/plugins:/home/straversaro/src/graphviz/build/install/lib
LS_COLORS=rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.Z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.jpg=01;35:*.jpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:
LC_TELEPHONE=it_IT.UTF-8
CLUTTER_BACKEND=x11
QT_ACCESSIBILITY=1
XDG_SESSION_PATH=/org/freedesktop/DisplayManager/Session0
GLADE_MODULE_PATH=:
XDG_SEAT_PATH=/org/freedesktop/DisplayManager/Seat0
SSH_AUTH_SOCK=/run/user/1001/keyring/ssh
DEFAULTS_PATH=/usr/share/gconf/xubuntu.default.path
SESSION_MANAGER=local/iiticublap103:@/tmp/.ICE-unix/2885,unix/iiticublap103:/tmp/.ICE-unix/2885
XDG_CONFIG_DIRS=/etc/xdg/xdg-xubuntu:/usr/share/upstart/xdg:/etc/xdg:/etc/xdg
GAZEBO_RESOURCE_PATH=/home/straversaro/src/gazebo-superbuild/build/install/share/gazebo-9::/home/straversaro/src/robotology-superbuild/build/install/share/gazebo/worlds
DESKTOP_SESSION=xubuntu
[... cut ...]
GAZEBO_MODEL_DATABASE_URI=http://gazebosim.org/models
LESSOPEN=| /usr/bin/lesspipe %s
INSTANCE=
UPSTART_JOB=startxfce4
CMAKE_PREFIX_PATH=:/home/straversaro/src/gazebo-superbuild/build/install:/home/straversaro/src/robotology-superbuild/build/install
XDG_RUNTIME_DIR=/run/user/1001
DISPLAY=:0.0
GLADE_CATALOG_PATH=:
XDG_CURRENT_DESKTOP=XFCE
GTK_IM_MODULE=
LESSCLOSE=/usr/bin/lesspipe %s %s
LC_TIME=it_IT.UTF-8
LC_NAME=it_IT.UTF-8
XAUTHORITY=/home/straversaro/.Xauthority
COLORTERM=xfce4-terminal
OLDPWD=/home/straversaro
_=/usr/bin/env
~~~

traversaro, 2018-05-18:
Running the command `idyntree-model-info -h` works fine?

anqingd, 2018-05-18:
The output of the ```ldd <location_where_you_saved_the_superbuild>/robotology-superbuild/build/install/lib/python2.7/dist-packages/_iDynTree.so```:
```
aduan@iiticublap086:~$ ldd /home/aduan/robotology-superbuild/build/install/lib/python2.7/dist-packages/_iDynTree.so
	linux-vdso.so.1 =>  (0x00007ffdbb371000)
	libidyntree-high-level-kdl.so => not found
	libidyntree-regressors.so => not found
	libidyntree-visualization.so => not found
	libpython2.7.so.1.0 => /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0 (0x00007fba0482a000)
	libidyntree-high-level.so => not found
	libidyntree-estimation.so => not found
	libidyntree-modelio-urdf.so => not found
	libidyntree-sensors.so => not found
	libidyntree-model.so => not found
	libidyntree-core.so => not found
	libstdc++.so.6 => /home/aduan/anaconda3/lib/libstdc++.so.6 (0x00007fba044f0000)
	libgcc_s.so.1 => /home/aduan/anaconda3/lib/libgcc_s.so.1 (0x00007fba042de000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fba03f14000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fba03cf7000)
	libz.so.1 => /home/aduan/anaconda3/lib/libz.so.1 (0x00007fba03ae0000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fba038dc000)
	libutil.so.1 => /lib/x86_64-linux-gnu/libutil.so.1 (0x00007fba036d9000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fba033d0000)
	/lib64/ld-linux-x86-64.so.2 (0x00007fba0522a000)
```

The output of the ```env``` command:
```
aduan@iiticublap086:~$ env
XDG_VTNR=7
LC_PAPER=en_US.UTF-8
LC_ADDRESS=en_US.UTF-8
XDG_SESSION_ID=c2
ROBOTOLOGY_SUPERBUILD_ROOT=/home/aduan/robotology-superbuild
XDG_GREETER_DATA_DIR=/var/lib/lightdm-data/aduan
LC_MONETARY=en_US.UTF-8
CLUTTER_IM_MODULE=xim
SESSION=ubuntu
GPG_AGENT_INFO=/home/aduan/.gnupg/S.gpg-agent:0:1
TERM=xterm-256color
VTE_VERSION=4205
XDG_MENU_PREFIX=gnome-
SHELL=/bin/bash
QT_LINUX_ACCESSIBILITY_ALWAYS_ON=1
GAZEBO_MODEL_PATH=/usr/share/gazebo-9/models:/home/aduan/icub-gazebo:/home/aduan/robotology-superbuild/build/install/share/gazebo/models
WINDOWID=71303178
LC_NUMERIC=en_US.UTF-8
UPSTART_SESSION=unix:abstract=/com/ubuntu/upstart-session/1001/1388
GNOME_KEYRING_CONTROL=
GTK_MODULES=gail:atk-bridge:unity-gtk-module
USER=aduan
LD_LIBRARY_PATH=:/home/aduan/mujoco/mjpro150/bin/:/usr/lib/x86_64-linux-gnu/gazebo-9/plugins
LS_COLORS=rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.Z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.jpg=01;35:*.jpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:
LC_TELEPHONE=en_US.UTF-8
QT_ACCESSIBILITY=1
XDG_SESSION_PATH=/org/freedesktop/DisplayManager/Session0
UNITY_HAS_3D_SUPPORT=true
XDG_SEAT_PATH=/org/freedesktop/DisplayManager/Seat0
SSH_AUTH_SOCK=/run/user/1001/keyring/ssh
DEFAULTS_PATH=/usr/share/gconf/ubuntu.default.path
XDG_CONFIG_DIRS=/etc/xdg/xdg-ubuntu:/usr/share/upstart/xdg:/etc/xdg
UNITY_DEFAULT_PROFILE=unity
MJLIB_PATH=/home/aduan/mujoco/mjpro150/bin/libmujoco150.so
GAZEBO_RESOURCE_PATH=/usr/share/gazebo-9::/home/aduan/robotology-superbuild/build/install/share/gazebo/worlds
DESKTOP_SESSION=ubuntu
PATH=/home/aduan/bin:/home/aduan/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/aduan/robotology-superbuild/build/install/bin
QT_IM_MODULE=ibus
QT_QPA_PLATFORMTHEME=appmenu-qt5
LC_IDENTIFICATION=en_US.UTF-8
XDG_SESSION_TYPE=x11
PWD=/home/aduan
[... 42 lines cut ...]
GTK_IM_MODULE=ibus
LESSCLOSE=/usr/bin/lesspipe %s %s
LC_TIME=en_US.UTF-8
LC_NAME=en_US.UTF-8
XAUTHORITY=/home/aduan/.Xauthority
_=/usr/bin/env

```

Running the command ```idyntree-model-info -h```:
```
aduan@iiticublap086:~$ idyntree-model-info -h
undefined short option: -h
usage: idyntree-model-info --model=string [options] ... 
options:
  -m, --model                         Model to load. (string)
  -p, --print                         Print the model.
      --total-mass                    Get the total mass of a model.
      --link-com                      Print inertial information of a given link (string [=])
      --link-com-frame                If link-com is passed, link-com-frame specifies the frame in which the inertia information is printed (assuming that the model is in zero position) (string [=])
      --frame-pose                    Print the referenceFrame_H_frame transform. (string [=])
      --frame-pose-reference-frame    If frame-pose is passed, frame-pose-reference-frame specifies the reference frame in which the frame pose is expressed (assuming that the model is in zero position) (string [=])
  -?, --help                          print this message

```

traversaro, 2018-05-18:
It seems that for some reason the [rpath](https://en.wikipedia.org/wiki/Rpath) settings of the bindings are not properly working. I think appending `/home/aduan/robotology-superbuild/build/install/` to [`LD_LIBRARY_PATH`](http://tldp.org/HOWTO/Program-Library-HOWTO/shared-libraries.html) could be a workaround to get the bindings to work.

anqingd, 2018-05-20:
Thanks @traversaro , the problem perhaps has been solved. I deleted anaconda and re-built robotology-superbuild as you suggested and finally ```import iDynTree``` works fine.

I am now trying to work the tutorial out especially the function ```setRobotState```:
https://github.com/robotology/idyntree/blob/61a07b20121333b6f230cc88623da864102db55e/examples/python/dynamicsComputationTutorial.py#L8. 
  

Attached are the logs with the iCubGenova04_plus model: 
```
('The loaded model has', 32, 'internal degrees of freedom and', 39, 'links.')
[WARNING]  :: parseURDFMaterial : Impossible to parse URDF material, material green not found in model database.
[WARNING]  :: parseURDFMaterial : Impossible to parse URDF material, material green not found in model database.
[WARNING]  :: parseURDFMaterial : Impossible to parse URDF material, material black not found in model database.
[WARNING]  :: parseURDFMaterial : Impossible to parse URDF material, material red not found in model database.
[WARNING]  :: parseURDFMaterial : Impossible to parse URDF material, material blue not found in model database.
[WARNING]  :: parseURDFMaterial : Impossible to parse URDF material, material yellow not found in model database.
[WARNING]  :: parseURDFMaterial : Impossible to parse URDF material, material pink not found in model database.
[WARNING]  :: parseURDFMaterial : Impossible to parse URDF material, material cyan not found in model database.
[WARNING]  :: parseURDFMaterial : Impossible to parse URDF material, material green not found in model database.
[WARNING]  :: parseURDFMaterial : Impossible to parse URDF material, material white not found in model database.
[WARNING]  :: parseURDFMaterial : Impossible to parse URDF material, material dblue not found in model database.
[WARNING]  :: parseURDFMaterial : Impossible to parse URDF material, material dgreen not found in model database.
[WARNING]  :: parseURDFMaterial : Impossible to parse URDF material, material gray not found in model database.
[WARNING]  :: parseURDFMaterial : Impossible to parse URDF material, material green not found in model database.
[WARNING]  :: parseURDFMaterial : Impossible to parse URDF material, material black not found in model database.
[WARNING]  :: parseURDFMaterial : Impossible to parse URDF material, material red not found in model database.
[WARNING]  :: parseURDFMaterial : Impossible to parse URDF material, material blue not found in model database.
[WARNING]  :: parseURDFMaterial : Impossible to parse URDF material, material yellow not found in model database.
[WARNING]  :: parseURDFMaterial : Impossible to parse URDF material, material pink not found in model database.
[WARNING]  :: parseURDFMaterial : Impossible to parse URDF material, material cyan not found in model database.
[WARNING]  :: parseURDFMaterial : Impossible to parse URDF material, material green not found in model database.
[WARNING]  :: parseURDFMaterial : Impossible to parse URDF material, material white not found in model database.
[WARNING]  :: parseURDFMaterial : Impossible to parse URDF material, material dblue not found in model database.
[WARNING]  :: parseURDFMaterial : Impossible to parse URDF material, material dgreen not found in model database.
[WARNING]  :: parseURDFMaterial : Impossible to parse URDF material, material gray not found in model database.
[WARNING]  :: parseURDFMaterial : Impossible to parse URDF material, material green not found in model database.
[WARNING]  :: parseURDFMaterial : Impossible to parse URDF material, material black not found in model database.
[WARNING]  :: parseURDFMaterial : Impossible to parse URDF material, material red not found in model database.
Traceback (most recent call last):
  File "/home/aduan/bullet3/data/icub-models/iCub/robots/iCubGenova04_plus/tryidyntree", line 28, in <module>
    dynComp.setRobotState(q, dq, ddq, gravity)
  File "/home/aduan/robotology-superbuild/build/install/lib/python2.7/dist-packages/iDynTree.py", line 7007, in setRobotState
    return _iDynTree.KinDynComputations_setRobotState(self, *args)
NotImplementedError: Wrong number or type of arguments for overloaded function 'KinDynComputations_setRobotState'.
  Possible C/C++ prototypes are:
    iDynTree::KinDynComputations::setRobotState(iDynTree::Transform const &,iDynTree::VectorDynSize const &,iDynTree::Twist const &,iDynTree::VectorDynSize const &,iDynTree::Vector3 const &)
    iDynTree::KinDynComputations::setRobotState(iDynTree::VectorDynSize const &,iDynTree::VectorDynSize const &,iDynTree::Vector3 const &)
```

traversaro, 2018-06-06:
@anqingd you were able to solve the problems? Can we close this issue?

anqingd, 2018-06-06:
Ah, I think the issue was solved and we can close this one.
Thanks! @traversaro

closed 2018-06-06 by anqingd