gbionics/idyntree #447: Drop support on Visual Studio 2013

traversaro, 2018-06-05:
I was wondering if in next iDynTree release (0.10) we can drop support for Visual Studio 2013 . This is what is blocking PR https://github.com/robotology/idyntree/pull/434 . I think this make sense because it is in line with YARP ( https://github.com/robotology/QA/issues/280 ).

However, I prefer to double check with @lucaTagliapietra and @diegoferigo if there is nothing strange (such as FT sensors drivers, VICON or Xsens drivers) that is forcing us to still use Visual Studio 2013 on some machine.

diegoferigo, 2018-06-06:
Nothing at my knowledge, let's get rid of it :tada:

traversaro, 2018-06-06:
I discussed f2f with @lucaTagliapietra and apparently the drivers of Xsens and the AMTI force plate require the  Microsoft Visual C++ Redistributable Packages for Visual Studio 2012 (or something similar, see https://support.microsoft.com/en-us/help/2977003/the-latest-supported-visual-c-downloads). 

However, once the Redistributable are installed, they can be used by code developed in Visual Studio 2015/2017, so it is perfectly ok to require VS2015 in iDynTree.