gbionics/idyntree #449: Enable tests that depend on IPOPT in Travis

traversaro, 2018-06-07:
In https://github.com/robotology/idyntree/pull/434 , a leftover of merge remained in the `InverseKinematics.h` . We did not noticed that because no build in Travis builds the part of iDynTree that depends on IPOPT. 

cc @S-Dafarra

S-Dafarra, 2018-06-07:
If I remember correctly, Travis is building with IPOPT. I had many problems in the past with IK related PRs :sweat_smile:. AppVeyor maybe is not compiling that part and the commit https://github.com/robotology/idyntree/pull/434/commits/5b0ae88fe4dbb72da324b784341763959b7adf14 was having ``[ci skip]`` in the message.

S-Dafarra, 2018-06-07:
Indeed Travis failed in the previous commit exactly for that error:
```
In file included from /home/travis/build/robotology/idyntree/src/inverse-kinematics/src/InverseKinematics.cpp:11:0:
/home/travis/build/robotology/idyntree/src/inverse-kinematics/include/iDynTree/InverseKinematics.h:663:1: error: expected unqualified-id before ‘<<’ token
 <<<<<<< 496d090c03bf452cd29c86cb84378ff160575525
 ^
In file included from /home/travis/build/robotology/idyntree/src/inverse-kinematics/src/InverseKinematics.cpp:11:0:
/home/travis/build/robotology/idyntree/src/inverse-kinematics/include/iDynTree/InverseKinematics.h:766:1: error: expected identifier before ‘<<’ token
 <<<<<<< 496d090c03bf452cd29c86cb84378ff160575525
 ^
/home/travis/build/robotology/idyntree/src/inverse-kinematics/include/iDynTree/InverseKinematics.h:766:1: error: expected ‘,’ or ‘...’ before ‘<<’ token
/home/travis/build/robotology/idyntree/src/inverse-kinematics/include/iDynTree/InverseKinematics.h:768:1: error: expected unqualified-id before ‘==’ token
 =======
 ^
/home/travis/build/robotology/idyntree/src/inverse-kinematics/include/iDynTree/InverseKinematics.h:770:1: error: expected unqualified-id before ‘>>’ token
 >>>>>>> Moved return type after deprecation to suppress warnings.
 ^
/home/travis/build/robotology/idyntree/src/inverse-kinematics/src/InverseKinematics.cpp:494:10: error: prototype for ‘bool iDynTree::InverseKinematics::setDesiredFullJointsConfiguration(const iDynTree::VectorDynSize&, double)’ does not match any in class ‘iDynTree::InverseKinematics’
     bool InverseKinematics::setDesiredFullJointsConfiguration(const iDynTree::VectorDynSize& desiredJointConfiguration, double weight)
          ^
In file included from /home/travis/build/robotology/idyntree/src/inverse-kinematics/src/InverseKinematics.cpp:11:0:
/home/travis/build/robotology/idyntree/src/inverse-kinematics/include/iDynTree/InverseKinematics.h:698:10: error: candidate is: bool iDynTree::InverseKinematics::setDesiredFullJointsConfiguration(const iDynTree::VectorDynSize&, const iDynTree::VectorDynSize&)
     bool setDesiredFullJointsConfiguration(const iDynTree::VectorDynSize& desiredJointConfiguration, const iDynTree::VectorDynSize& weights);
          ^
/home/travis/build/robotology/idyntree/src/inverse-kinematics/src/InverseKinematics.cpp:652:10: error: prototype for ‘void iDynTree::InverseKinematics::getSolution(iDynTree::Transform&, iDynTree::VectorDynSize&)’ does not match any in class ‘iDynTree::InverseKinematics’
     void InverseKinematics::getSolution(iDynTree::Transform & baseTransformSolution,
          ^
In file included from /home/travis/build/robotology/idyntree/src/inverse-kinematics/src/InverseKinematics.cpp:11:0:
/home/travis/build/robotology/idyntree/src/inverse-kinematics/include/iDynTree/InverseKinematics.h:765:10: error: candidate is: void iDynTree::InverseKinematics::getSolution(iDynTree::Transform&, int)
     void getSolution(iDynTree::Transform& baseTransformSolution,
          ^
/home/travis/build/robotology/idyntree/src/inverse-kinematics/src/InverseKinematics.cpp:660:90: error: no ‘void iDynTree::InverseKinematics::getFullJointsSolution(iDynTree::Transform&, iDynTree::VectorDynSize&)’ member function declared in class ‘iDynTree::InverseKinematics’
                                                   iDynTree::VectorDynSize & shapeSolution)
                                                                                          ^
make[2]: *** [src/inverse-kinematics/CMakeFiles/idyntree-inverse-kinematics.dir/src/InverseKinematics.cpp.o] Error 1
make[1]: *** [src/inverse-kinematics/CMakeFiles/idyntree-inverse-kinematics.dir/all] Error 2
make: *** [all] Error 2
```

traversaro, 2018-06-07:
Ok, then it is simply my fault! That's better. Closing the issue.

closed 2018-06-07 by traversaro

traversaro, 2018-06-07:
There was also another failure: https://github.com/robotology/idyntree/commit/821aa5f1791be12deeeaf0f2fff7ff7de5432953 . 

In a nutshell, using `[ci skip]` in the last commit of a PR is simply crazy, my bad.