gbionics/idyntree #451: Compilation of iDynTree devel against yarp devel

diegoferigo, 2018-06-12:
Wb-toolbox's Travis encountered the following issue while compiling iDynTree devel branch against Yarp new devel branch:

```
$ cmake .. \
  -G"$TRAVIS_CMAKE_GENERATOR" \
  -DCMAKE_BUILD_TYPE=$TRAVIS_BUILD_TYPE \
  -DCMAKE_INSTALL_PREFIX=$DEPS_CACHE
-- The CXX compiler identification is GNU 4.8.4
-- Check for working CXX compiler: //home/travis/.local/bin/g++
-- Check for working CXX compiler: //home/travis/.local/bin/g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Eigen3: /usr/include/eigen3 (Required is at least version "3.2.92") 
-- Using iCub from install
CMake Warning at cmake/iDynTreeDependencies.cmake:14 (find_package):
  Could not find a configuration file for package "YARP" that is compatible
  with requested version "2.3.62".
  The following configuration files were considered but not accepted:
    /home/travis/cache/linux/Release/lib/cmake/YARP/YARPConfig.cmake, version: 3.0.100
Call Stack (most recent call first):
  cmake/iDynTreeDependencies.cmake:44 (idyntree_handle_dependency)
  CMakeLists.txt:43 (include)
-- Found IPOPT: /usr/lib/libipopt.so;/usr/lib/x86_64-linux-gnu/libdl.so;/usr/lib/libcoinmumps.so;/usr/lib/x86_64-linux-gnu/libpthread.so;/usr/lib/libcoinlapack.so;/usr/lib/libblas.so;/usr/lib/gcc/x86_64-linux-gnu/4.8/libgfortran.so;/usr/lib/x86_64-linux-gnu/libm.so;/usr/lib/gcc/x86_64-linux-gnu/4.8/libquadmath.so;/usr/lib/libcoinmetis.so  
-- Found TinyXML: /usr/lib/x86_64-linux-gnu/libtinyxml.so  
CMake Error at src/core/CMakeLists.txt:131 (target_compile_features):
  target_compile_features The compiler feature "cxx_attribute_deprecated" is
  not known to CXX compiler
  "GNU"
  version 4.8.4.
-- The C compiler identification is GNU 4.8.4
-- Check for working C compiler: //home/travis/.local/bin/gcc
-- Check for working C compiler: //home/travis/.local/bin/gcc -- works
```

cc @traversaro

traversaro, 2018-06-12:
Fixed by https://github.com/robotology/idyntree/pull/452 , feel free to review.

diegoferigo, 2018-06-12:
Super quick, thanks! Looks good to me

traversaro, 2018-06-12:
Yep, this is a result of dropping support for Ubuntu 14.04 and GCC 4 (see https://github.com/robotology/QA/issues/173). What do you think it would be a good idea here? An improved error message?

diegoferigo, 2018-06-12:
There is already the error in CMake output (read the errors in the first post), but this breaks **all** our tests in Travis, not due to c++11 but due to c++14 requirements.

diegoferigo, 2018-06-12:
If this is the case (I'm still checking), what we should do is just update the compiler in our CI pipeline if the support to GCC < 5 has already been dropped. We're already running on a unsupported version.

traversaro, 2018-06-12:
See https://github.com/robotology/idyntree/pull/434#issuecomment-382153001 for the related discussion. For Travis I did exactly that following the YARP example, see https://github.com/robotology/idyntree/blob/devel/.travis.yml#L82 . I am not aware anyhow of any project that currently depends on iDynTree devel, so I think we can fix the problem before the release so we make sure not to break any Travis build.

diegoferigo, 2018-06-12:
Emh now wb-toolbox devel uses idyntree devel, this led me here :sweat_smile: I will port that workaround also on wb-toolbox's Travis, good to know :+1: Thanks!

closed 2018-06-12 by diegoferigo