gbionics/idyntree #453: Release iDynTree 0.10

traversaro, 2018-06-13:
The release of some features available in `devel` is blocking https://github.com/robotology/robotology-superbuild/pull/87 . 

Before releasing iDynTree 0.10, we should check there is not blocking the release in any way.

For sure @diegoferigo already noticed that iDynTree 0.10 is not compatible with GCC 4 (this was actually intentional), so releasing this version will break all our Travis builds that use GCC 4 (the default compiler in Travis). Using GCC 5 in Travis is possible (see https://github.com/robotology/idyntree/blob/devel/.travis.yml#L16 and https://github.com/robotology/idyntree/blob/devel/.travis.yml#L83), but we should do this change on all the repositories that actually build iDynTree in their Travis script.

traversaro, 2018-06-13:
Repos that should update their travis to Gcc 5 : 
 * https://github.com/robotology/human-dynamics-estimation
 * https://github.com/robotology/wb-toolbox/blob/master/.travis.yml

traversaro, 2018-06-13:
@diegoferigo @nunoguedelha @DanielePucci @S-Dafarra @fjandrad  @gabrielenava @lucaTagliapietra @claudia-lat @hu-yue @aikolina @jeljaik @prashanthr05  any idea on something that is blocking this release?

diegoferigo, 2018-06-14:
@traversaro You listed already the repositories I knew they need an update. At first glance I am not aware of any other project which might be affected by this release.

Since the change (at least in Travis) is trivial, I would proceed with the release.

traversaro, 2018-06-17:
Another repo that needs to be updated is https://github.com/robotology/icub-model-generator . 

> Since the change (at least in Travis) is trivial, I would proceed with the release.

Feel free to do the change, it should work also with the old version of iDynTree .

traversaro, 2018-06-20:
icub-model-generator fixed in https://github.com/robotology/icub-model-generator/issues/94 , I think we can proceed with the release.

diegoferigo, 2018-06-20:
Go ahead, I will fix wb-toolbox's Travis soon.

traversaro, 2018-06-21:
Fixed in https://github.com/robotology/idyntree/pull/455 .

closed 2018-06-21 by traversaro