gbionics/idyntree #461: Investigate why SWIG does not like `[[deprecated]]` attributes

traversaro, 2018-06-28:
We now require C++14 to be used in the header, so in theory we could get rid of `IDYNTREE_DEPRECATED_WITH_MSG` and `IDYNTREE_DEPRECATED` in a favor of `[[deprecated]]` attributes. However SWIG seems to still have problem with it, we should check if it is a problem in old version of SWIG and eventually get rid of `IDYNTREE_DEPRECATED_WITH_MSG` and `IDYNTREE_DEPRECATED` .

francesco-romano, 2018-06-28:
[here](https://github.com/swig/swig/issues/490) they discuss about implementing the parsing of the attribute. I do not know if this means that SWIG is not able to ignore it if found.

traversaro, 2018-09-10:
Apparently `SWIG 3.*` handles it correctly (i.e. it does not crash), while SWIG 2.0 has problems.

necil, 2018-12-14:
> Apparently `SWIG 3.*` handles it correctly (i.e. it does not crash), while SWIG 2.0 has problems.

I have version 3.0.10-1.1 and it fails while generating the interface.

traversaro, 2018-12-14:
Hi @necil, what is the precise error that you are getting? It is related to `[[deprecated]]`?

necil, 2018-12-14:
Hi @traversaro, thanks for support and quick reply.
It faıls when I add [[deprecated]] and error diessappears when I remove it from code. Error consists of `Error: Syntax error in input(3)` and line in code.
I have 2 questions: First one do you suggest a way to see more detailed log from this error? and and second one should I add a directive on my swig interface file to enable usage of deprecated?

traversaro, 2018-12-14:
Just to clarify: you are discussing about a separate C++ project on which you run SWIG, not about iDynTree, right?

traversaro, 2018-12-14:
> I have 2 questions: First one do you suggest a way to see more detailed log from this error? and and second one should I add a directive on my swig interface file to enable usage of deprecated?

Can you build a Minimal Working Example reproducing this problem?

wangdf, 2020-05-07:
http://www.swig.org/Doc4.0/SWIGDocumentation.html#CPlusPlus11_attributes

necil, 2020-05-08:
@traversaro yeah it was a dummy code to just test that deprecate directive works. To be honest I do not remember why I have commented it under this repository it was a while ago. But probably it was a fault on my side.