gbionics/idyntree #464: Compute partial derivatives of jacobian w.r.t. to joint positions

traversaro, 2018-07-04:
@S-Dafarra asked about this. This derivative have the nice properties that everything boils down to a lot of 6D cross product between all the joint motion subspace vectors. 

References on this: 
* Equation 31 of https://www.researchgate.net/profile/Gianluca_Garofalo2/publication/259899572_On_the_closed_form_computation_of_the_dynamic_matrices_and_their_differentiations/links/59707389a6fdccc6c973a8e1/On-the-closed-form-computation-of-the-dynamic-matrices-and-their-differentiations.pdf 
* Equation 13 of https://ieeexplore.ieee.org/stamp/stamp.jsp?tp=&arnumber=240619 

Also check the iKin implementation of Hessians: https://github.com/robotology/icub-main/blob/master/src/libraries/iKin/src/iKinFwd.cpp#L1124 .

S-Dafarra, 2018-07-04:
What about the floating base part? Should we consider a 6dof joint there?

S-Dafarra, 2018-07-04:
Just to provide another source, Equation 13 of https://ieeexplore.ieee.org/stamp/stamp.jsp?tp=&arnumber=240619 corresponds to Equation 37 of https://ieeexplore.ieee.org/stamp/stamp.jsp?arnumber=1554272

traversaro, 2018-07-04:
It depends what you need to compute for the floating base part. Depending on the choice of the frame velocity representation, the joint part of the jacobian may or may not depend on the floating base position/orientation, while for the derivatives of the adjoint transform (i.e. that may be the base part of the jacobian, again depending on the velocity representation) the procedure is a bit more complicated. 
If you write down formally the quantity you want to derive, I think it will be easier to explain.
 For the robot momentum it is a bit more complicated because you have to account for all the composite rigid body inertia present in the formula of the momentum jacobian.

traversaro, 2018-07-04:
FYI @mpbos This is a bit hard to understand know, but it is definitely related to your master thesis work.

traversaro, 2018-07-04:
Related code in Pinocchio: 
* https://github.com/stack-ofhttps://github.com/dartsim/dart/blob/071ead1bc29bacf28e39c9a88ffe9270baf072eb/dart/dynamics/FixedJacobianNode.cpp#L192-tasks/pinocchio/blob/v1.2.9/bindings/python/scripts/derivative/dcrba.py 

Related code in DART:
* https://github.com/dartsim/dart/blob/a206b1995a4c4ee96de691351cb109b808d0a347/dart/dynamics/BodyNode.cpp#L2398

Related code in KDL: 
* https://github.com/orocos/orocos_kinematics_dynamics/blob/master/orocos_kdl/src/chainjnttojacdotsolver.cpp#L117

S-Dafarra, 2018-07-04:
Aren't those related to time derivative? :thinking:

traversaro, 2018-07-04:
Yep,  but if you expand the use of any velocity with $\mathrm{v} = J \nu$, you almost obtain the partial derivatives w.r.t. to the joint positions.

jcarpent, 2018-07-04:
Dear iDynTree developers,

I have implemented such kind of derivatives here:
https://github.com/stack-of-tasks/pinocchio/blob/devel/src/algorithm/kinematics-derivatives.hxx

I never explicitly compute the tensor H of dimension 6 x nv x nv (where nv is the dimension of the tangent vector - aka joint velocity) but rather H*v (v being the joint velocity).
If you have any question on the code, I will be pleased to help you.

Best,

Justin

traversaro, 2018-07-04:
Thanks @jcarpent for the useful insight and references.

S-Dafarra, 2018-07-04:
Thanks @jcarpent! I will definitely go though that. I updated the [above comment](https://github.com/robotology/idyntree/issues/464#issuecomment-402405375) to have the references all together.

By any chance, do you have any document about the mathematical derivation you implemented?

jcarpent, 2018-07-04:
A bit is explained here:
https://hal.laas.fr/hal-01790971
When deriving RNEA derivatives, you can get for free the derivatives of the kinematics too.
I will try to make these things clearer in a companion document that I need to write.

If you have any questions, I will be pleased to answer them.

S-Dafarra, 2018-07-04:
Thanks again!

traversaro, 2019-08-29:
Related issue in pinocchio: https://github.com/stack-of-tasks/pinocchio/issues/867 .