gbionics/idyntree #467: Release 0.11

traversaro, 2018-08-20:
We have a few fixes just in `devel` that it would be nice to have in master, and  some new features such as the new XML parser that it would be nice to release. I think we can proceed to release 0.11 relatively soon, any comment? 
@diegoferigo @DanielePucci @prashanthr05

ahoarau, 2018-08-20:
maybe #383 :dancer:

traversaro, 2018-08-20:
Ok!

traversaro, 2018-08-22:
I think this can also be the last release compatible with YARP 2.3.72 .

traversaro, 2018-08-30:
Sorry @ahoarau , we are kind in a hurry and I am going to release 0.11 now. I hope to get https://github.com/robotology/idyntree/issues/383 in 0.12 !

ahoarau, 2018-08-30:
@traversaro No prob. Could you also mention the c++ version you use ? 11 ? 14 ?

traversaro, 2018-08-30:
Good point. Since this release, we use C++14 in the headers, I will add that on the release notes.

traversaro, 2018-09-03:
Done.

closed 2018-09-03 by traversaro