gbionics/idyntree #485: When generating HDE iDynTree::idyntree-high-level includes non-existant

mebbaid, 2018-09-14:
From the title, when I was configuring and generating the solutions for the HDE , the include dirs for the `iDynTree::idyntree-high-level` which the `CMakeLists.txt` of human-forces-provider projecttries to link against leads to a non-existing path. Below is a screenshot of the error.

[image]

mebbaid, 2018-09-14:
No , The work around that @diegoferigo  suggested was to create the empty private directory in the include dirs like in the attached picture (untill the cmake issue is found out):

[image]

traversaro, 2018-09-14:
Sorry @mebbaid , the problem was more widespread than I expected: this new commit  https://github.com/robotology/idyntree/pull/486/commits/0877c0359a1e5c19ff5ffceb3e63c77cf10f8828 should solve the issue. Can you update the branch and try again?

traversaro, 2018-09-27:
@mebbaid post-demo, do you have any idea if this was solved or not?

mebbaid, 2018-09-28:
Not working for me yet!

traversaro, 2018-09-28:
New build or old build? Latest `master`?

mebbaid, 2018-09-28:
YARP is `HEAD detached at 4d82b1e70`
everything else of the `superbuild` is on `master`. The wearables is on `origin/feature/cleanup`

traversaro, 2018-09-28:
> everything else of the superbuild is on master.

iDynTree has been updated  to the latest master?

mebbaid, 2018-09-28:
Probably that's the mistake. let me check