gbionics/idyntree #494: How is the mass matrix built for the floating base?

vpadois, 2018-09-19:
Hello,

working with @ahoarau, @rlober and @paulinejmaurice on ORCA (https://github.com/syroco/orca) which rely on iDynTree for model computations, we are wondering how is the mass matrix built for the floating base. How is the mass and inertia of the root body distributed among the different free floating joints ?

Thanks in advance
Vincent

traversaro, 2018-09-20:
Hi @vpadois ! 
We are currently improving the documentation, but some good references for the structure of the mass matrix as returned by the KinDynComputatiions, if the KinVelRapresentation is `MIXED_REPRESENTATION` in iDynTree
* Equation 70 of https://research.tue.nl/files/25753352/Traversaro_en_Saccon_DC_2016.064.pdf  
* Equation 13 of https://www.researchgate.net/profile/Daniele_Pucci/publication/312200239_A_Unified_View_of_the_Equations_of_Motion_used_for_Control_Design_of_Humanoid_Robots/links/5878dd9708aed3826ae1d3a3/A-Unified-View-of-the-Equations-of-Motion-used-for-Control-Design-of-Humanoid-Robots.pdf

traversaro, 2018-09-20:
>  Does that mean that iDynTree does not rely on a serial chain description of the free-floating joints (as that was the case for some humanoids in some old xml file I used to play with long time ago)? If yes that's great and I do not have more question on this for the time being.

Exactly, we model **explicitly** the floating base position/velocity/acceleration, and that's why in most method for setting/getting the state of the robot we pass explicitly the position/velocity/acceleration of the base.

vpadois, 2018-09-20:
Thanks for the confirmation.

closed 2018-09-20 by vpadois