gbionics/idyntree #496: Parsing a URDF <visual> or <collision> with the origin tag produces an invalid `link_H_geometry` transform

traversaro, 2018-09-27:
If you parse a element such as:
~~~xml
   <collision>
     <geometry>
       <mesh filename="cube.stl"/>
     </geometry>
   </collision>
~~~
the resulting `link_H_geometry` in the shape object is not initialized. If you just add a origin;
~~~xml
   <collision>
     <origin xyz="0 0 0" rpy="0 0 0" />
     <geometry>
       <mesh filename="cube.stl"/>
     </geometry>
   </collision>
~~~
everything works fine.

francesco-romano, 2018-09-27:
Because `iDynTree::Transform` does not initialise to identity by default.
Should be enough to either
- add a constructor to https://github.com/robotology/idyntree/blob/master/src/model_io/urdf/include/private/VisualElement.h#L37 or
- initialise it here: https://github.com/robotology/idyntree/blob/master/src/model_io/urdf/src/VisualElement.cpp#L26

traversaro, 2018-09-27:
> Because `iDynTree::Transform` does not initialise to identity by default.

Yes, that was a controversial decision but it proved to be necessary for performance (see https://github.com/robotology/idyntree/issues/98). Thanks for the quick suggestion!

francesco-romano, 2018-09-27:
Yes, I remember that decision. But I always forget about it. During tests I caught a couple of these errors, but I think I missed this one.

I think it would be nice to run again #98 with more modern compiler and C++ stuff (like `= default`) to see how things have changed during time.

ahoarau, 2019-08-26:
Anyway we can have a demo on how to use/get/read/dostuff with the collision and visual elements ?

traversaro, 2019-08-26:
> Any way we can have a demo on how to use/get/read/dostuff with the collision and visual elements ?

Thanks for the suggestion, I opened a new issue at https://github.com/robotology/idyntree/issues/563 . Not sure if I can work on it in the next future, in the mean while there are some existing places in the codebase in which the API is used: 
* https://github.com/robotology/idyntree/blob/c8bf721b771fa4b1e7c3a940632e121060719a19/src/visualization/src/ModelVisualization.cpp#L91 
* https://github.com/robotology/idyntree/blob/c8bf721b771fa4b1e7c3a940632e121060719a19/src/model/src/ModelTransformers.cpp#L300

closed 2019-08-27 by traversaro