gbionics/idyntree #519: Compilation error: iDynTree/Estimation/AttitudeEstimatorUtils.h: No such file or directory

prashanthr05, 2019-01-29:
I am trying to compile a project that implements the `AttitudeMahonyFilter` and `AttitudeQuaternionEKF` classes. However, I get the following compilation error while building the project
``` make
[ 36%] Built target floatingBaseEstimationRPC-service
[ 45%] Building CXX object CMakeFiles/icubFloatingBaseEstimatorV1.dir/yarp_plugin_icubFloatingBaseEstimatorV1.cpp.o
In file included from /home/pramadoss/iit_ws/loc2/element_floating-base-estimation/code/icubFbeV1/include/icubFloatingBaseEstimatorV1.h:17:0,
                 from /home/pramadoss/iit_ws/loc2/element_floating-base-estimation/code/icubFbeV1/build/yarp_plugin_icubFloatingBaseEstimatorV1.cpp:13:
/home/pramadoss/iit_ws/robotology-superbuild/build/install/include/iDynTree/Estimation/AttitudeMahonyFilter.h:15:56: fatal error: iDynTree/Estimation/AttitudeEstimatorUtils.h: No such file or directory
compilation terminated.
CMakeFiles/icubFloatingBaseEstimatorV1.dir/build.make:62: recipe for target 'CMakeFiles/icubFloatingBaseEstimatorV1.dir/yarp_plugin_icubFloatingBaseEstimatorV1.cpp.o' failed
make[2]: *** [CMakeFiles/icubFloatingBaseEstimatorV1.dir/yarp_plugin_icubFloatingBaseEstimatorV1.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/icubFloatingBaseEstimatorV1.dir/all' failed
make[1]: *** [CMakeFiles/icubFloatingBaseEstimatorV1.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
```

This maybe due to fact that `AttitudeEstimatorUtils.h` is defined as private include in [iDyntree/estimation/CMakelists](https://github.com/robotology/idyntree/blob/1829e1dc3d90a8f6fb289bfb0ab5a16e8d9adc9b/src/estimation/CMakeLists.txt#L25). However, I am not trying to directly access this header through the project `icubFloatingBaseEstimatorV1`, but it is being included through the [AttitudeMahonyFilter.h](https://github.com/robotology/idyntree/blob/1829e1dc3d90a8f6fb289bfb0ab5a16e8d9adc9b/src/estimation/include/iDynTree/Estimation/AttitudeMahonyFilter.h#L15)  file.

Any suggestions on how to handle this compilation error would be really helpful. Thanks in advance.

traversaro, 2019-01-29:
Sorry, that was an error of mine during the review. Private headers cannot be include in private headers, exactly for the problem that you are facing. In this case, it sufficient to move the include in the `AttituteMahonyFilter.cpp` file.  In general, for reducing compilation times is typically beneficial to just include the strictly necessary headers in the public headers (sometimes using forward declaration).

traversaro, 2019-01-30:
> You mean to say private headers cannot be included in public headers ?

Sorry, just corrected the post.

prashanthr05, 2019-02-04:
Fixed in https://github.com/robotology/idyntree/commit/ad1f70c63f5f38433cccbb7360b84317be929bc5.

closed 2019-02-04 by prashanthr05