gbionics/idyntree #523: Error in parsing URDF model if missing a new line at the end of the .urdf

claudia-lat, 2019-02-27:
Today, when I tried to load a .urdf model with the `loadModelFromFile` method via Matlab bindings, it gave me this error:

```matlab
test = iDynTree.ModelLoader();
test.loadModelFromFile('model.urdf','urdf')
[ERROR]  :: modelFromURDFString : Invalid XML
[ERROR] ModelLoader :: loadModelFromFile : Error in parsing model from URDF.
```

By investigating the problem I found it is related to the presence (or not) of the new line at the end of the .urdf, such that:

- **if there's no new line, it fails**

[image]

- **if there is the line, it works**

[image]

Important notes: 
1. the error occurred in mac OS, with Matlab 2016b
2. test on Ubuntu, Matlab 2018 --> no error occurred
3. test on mac OS, with Matlab 2017 --> no error occurred

diegoferigo, 2019-02-28:
This was happening also using `idyntree-model-info`, raising the same error. I suspect that it is something related to the urdf parser.

cc @traversaro @francesco-romano

traversaro, 2019-02-28:
Related issue: https://github.com/robotology/idyntree/issues/243 .

> * the error occurred in mac OS, with Matlab 2016b
> * test on Ubuntu, Matlab 2018 --> no error occurred
> * test on mac OS, with Matlab 2017 --> no error occurred

Which version of Ubuntu/macOS? I suspect the difference in behaviour is due to the libxml2 version.

claudia-lat, 2019-02-28:
> Which version of Ubuntu/macOS?
* the error occurred in mac OS **High Sierra, ver 10.13.6**, with Matlab 2016b 
* test on Ubuntu **18.04**, Matlab 2018 --> no error occurred
* test on mac OS **Sierra, ver 10.12.6**, with Matlab 2017 --> no error occurred

traversaro, 2019-02-28:
I suspect Matlab ships a libxml2 itself.

diegoferigo, 2019-02-28:
> I suspect Matlab ships a libxml2 itself

Yes it does.

> This was happening also using `idyntree-model-info`, raising the same error. I suspect that it is something related to the urdf parser.

However, this would just explain the behaviour when idyntree bindings are called from Matlab. Since this is happening also calling  `idyntree-model-info`, I tend to assume that this issue is unrelated to the usage of Matlab.

traversaro, 2019-02-28:
> However, this would just explain the behaviour when idyntree bindings are called from Matlab. Since this is happening also calling  `idyntree-model-info`, I tend to assume that this issue is unrelated to the usage of Matlab.

The issue with `idyntree-model-info` is only happening on mac OS **High Sierra, ver 10.13.6** as well?

DanielePucci, 2019-03-02:
CC
@claudia-lat

claudia-lat, 2019-03-03:
The error with `idyntree-model-info` doesn't appear with Ubuntu. To be tested with mac OS Sierra, ver 10.12.6 (cc @nunoguedelha).