gbionics/idyntree #526: Significant error in Gravity Torque

VenusPasandi, 2019-04-04:
I used the 'iCubGazeboV2_5 (no hands)' in the gazebo and control its position with 'set position'. Then, I compared the 'joint torques' and 'Gravity Torques' to check the gravity compensation. As you imagine, the Simulink model is simple as
[image]
The result is as
[image]
There is a significant error between the gravity force and the joint torques!

DanielePucci, 2019-04-04:
@VenusPasandi which joint were you looking at?

VenusPasandi, 2019-04-05:
@DanielePucci ,
The results reported in the previous comment is for `l_elbow`. Though, I also tested `l_shoulder_roll`,`r_shoulder_roll`,`r_elbow`.

traversaro, 2019-04-05:
Let's move this issue in WB-Toolbox.