gbionics/idyntree #527: Update yarprobotstate publisher joints check

Yeshasvitvs, 2019-04-10:
Currently, the `yarprobotstatepublisher` [checks](https://github.com/robotology/idyntree/blob/b127441bd34342898797c09db7ce1292a2cfe7fe/src/tools/yarprobotstatepublisher/src/robotstatepublisher.cpp#L202) for the **exact match** between the number of joints present in the urdf model passed to `--model` parameter and the number of joints present in the `/joints_states` topic. Ideally, it is better to update this check to ensure if the `/joint_states` topic has at least the same number of joints as that of the urdf model. This avoids the problem where the urdf model used will generally have around 32 joints without considering  the hand joints, whereas the real robot streams the `/joint_states` topic along with the hand joints through the [configuration file](https://github.com/robotology/robots-configuration/blob/ff1e768f2856a3fac2d82c05d4d11d1b33bb3e52/iCubGenova04/wrappers/motorControl/icub_ros_wrapper.xml#L9)

Yeshasvitvs, 2019-04-10:
@traversaro Can I proceed to change this in the branch with the PR https://github.com/robotology/idyntree/pull/513

traversaro, 2019-04-10:
Yes, please, go on to modify the branch used in that PR.

Yeshasvitvs, 2019-04-11:
@traversaro I do not have push access to upstream repo

diegoferigo, 2019-04-13:
Waiting for @traversaro you can open a PR that targets that branch

Yeshasvitvs, 2019-04-13:
Opened a PR https://github.com/robotology/idyntree/pull/529

Yeshasvitvs, 2019-04-16:
PR merged, closing this issue.

traversaro, 2019-04-16:
Hi @Yeshasvitvs , we tipically close issues when the problem is solved at least in the `devel` branch, I think we should keep the issue open until https://github.com/robotology/idyntree/pull/513 is merged.

lrapetti, 2019-05-11:
We should allow also the fact that the message has less joint that the actual number of joint of the model, in this case, just the model for which we have the message would be updated.
I think @Yeshasvitvs has already implemented a preliminary version of those changes and he may add them in the PR.

traversaro, 2019-05-11:
>  I think @Yeshasvitvs has already implemented a preliminary version of those changes and he may add them in the PR.

First of all, thanks for providing those fixes. However, given that the PR https://github.com/robotology/idyntree/pull/513 has been open for quite a long time and already contains a lot of different features, I would suggest to get it merged without adding new feature to it. Once it is merged, we can add new features with a new PR.

traversaro, 2020-01-18:
Fixed in  https://github.com/robotology/idyntree/pull/551 .

closed 2020-01-18 by traversaro