gbionics/idyntree #528: Implement inverse velocity kinematic solver

lrapetti, 2019-04-11:
I am currently working on the implementation of a solver for inverse-velocity-kinematics in view of implementing velocity integration based inverse kinematics solver.

The implementation can be found in [`lrapetti/idyntree/feature/inverse-velocity-kinematics`](https://github.com/lrapetti/idyntree/tree/feature/inverse-velocity-kinematics).

@traversaro @S-Dafarra @GiulioRomualdi @Yeshasvitvs

traversaro, 2019-04-11:
I added a few comments in the commits. In general  take everything you find in the existing inverse kinematics class with a grain of salt, see https://github.com/robotology/idyntree/issues/282 .

lrapetti, 2019-04-11:
> In general take everything you find in the existing inverse kinematics class with a grain of salt, see #282

If it's the case, I will try to port some of the improvements we may discuss for `inverse-velocity-kinematics` to the `inverse-kinematics` class with small fixies.