gbionics/idyntree #540: Adding RPYRightTrivializedSecondDerivative() into iDyntree::Rotation class

GiulioRomualdi, 2019-06-12:
Checking `iDynTree` documentation seems that the second derivative of `RPYRightTrivialized` matrix is not implemented (see [here](http://wiki.icub.org/codyco/dox/html/idyntree/html/classiDynTree_1_1Rotation.html#ade94c567710848fe366c1554279f3c1e) for the first derivative). 
This new method can be useful for evaluating the angular acceleration from the Euler angle rate of change.

If you are interested I can propose a PR with the implementation 

cc @MiladShafiee @traversaro

traversaro, 2019-06-12:
Yes I think it would make sense to have it.

GiulioRomualdi, 2019-06-13:
Related PR #542

traversaro, 2019-06-14:
Thanks, fixed in https://github.com/robotology/idyntree/pull/542 .

PR #542 (2019-06-13): Implement RPYRightTrivializedDerivativeRateOfChange() and RPYRightTrivializedDerivativeInverseRateOfChange() into Rotation class
This PR implement `RPYRightTrivializedDerivativeRateOfChange()` and `RPYRightTrivializedDerivativeInverseRateOfChange()` methods into `Rotation` class. 

I also implemented the test for testing the aforementioned methods.  

Close #540 

cc @MiladShafiee @S-Dafarra @traversaro
closed 2019-06-14 by traversaro