gbionics/idyntree #546: [estimation] Possible error in SimpleLeggedOdometry init method computation

prashanthr05, 2019-07-08:
The `init()` method of the `SimpleLeggedOdometry` class  seems ambiguous in the sense that it takes as an input argument `iDynTree::FrameIndex` but carries out the forward kinematics computation only up to the associated link of the input frame.

https://github.com/robotology/idyntree/blob/ed3e020ee38a2b65c08ac5e48454a24f2944197c/src/estimation/src/SimpleLeggedOdometry.cpp#L50

This might cause confusion to the users who might be passing, for example, a frame called `l_sole` as an `initialFixedFrame` while the computations are done only for `l_foot`, which might cause propagation of error in the odometry.
``` cpp
    m_fixedLinkIndex = m_model.getFrameLink(initialFixedFrameIndex);

    Transform world_H_initialFixedFrame = initialFixedFrame_H_world.inverse();
    Transform initalFixedFrame_H_fixedLink =  m_model.getFrameTransform(m_fixedLinkIndex).inverse();

    m_world_H_fixedLink = world_H_initialFixedFrame*initalFixedFrame_H_fixedLink;
```
https://github.com/robotology/idyntree/blob/ed3e020ee38a2b65c08ac5e48454a24f2944197c/src/estimation/src/SimpleLeggedOdometry.cpp#L73

Possibly related issue 
#219

prashanthr05, 2019-07-08:
I understand that it is just a level of abstraction and will not cause any errors.

Sorry for the noise! ;D Closing the issue.

closed 2019-07-08 by prashanthr05

traversaro, 2019-07-29:
I am not really sure to understand your initial concern. Basically, if a frame is known to be fixed, we infer that the link rigidly connected to that frame will be fixed as well, and this is what we assume in that method. Do you think that makes sense?

prashanthr05, 2019-07-29:
What I actually meant was that the input to the method is an arbitrary frame rigidly attached to the link, while internally the forward kinematics along the Traversal chain is computed only until the link and not to the frame itself. So I thought probably this constant link_H_frame is being ignored every time the LeggedOdometry is updated. 

However, I noted that we do not bring the frame into the computations at all, and only deal with the associated link. 

which corroborates with your statement
> we infer that the link rigidly connected to that frame will be fixed as well, and this is what we assume in that method

prashanthr05, 2020-06-15:
There was indeed an error in the init method computation. See https://github.com/robotology/idyntree/pull/698.