gbionics/idyntree #550: [yarprobotstatepublisher] The BottleImpl error flood

prashanthr05, 2019-08-02:
Whenever we run the `yarprobotstatepublisher` the following error floods the terminal

``` bash
yarp: BottleImpl reader failed, unrecognized object code 0
```

Any suggestions to fix this?

traversaro, 2019-08-03:
I have no idea why this happen. Have you tried to run a debugger to understand which line generates this error message?

traversaro, 2020-01-08:
@prashanthr05 Are you still experiencing this? Do you know how to reproduce it?

prashanthr05, 2020-01-09:
Unfortunately, I had not gone back to addressing/testing/reproducing this problem. Maybe we could close this issue at the moment, and reopen if this issue arises again.

traversaro, 2020-01-09:
> Unfortunately, I had not gone back to addressing/testing/reproducing this problem. Maybe we could close this issue at the moment, and reopen if this issue arises again.

Ok!

closed 2020-01-09 by traversaro