gbionics/idyntree #581: Benchmark / Profiling

ahoarau, 2019-09-18:
Hello iDyntree people,
I think it would be beneficial to give users a simple example they can run so they have an idea of how much time each function cost in term of milliseconds:
- For a simple robot arm (6 dof), forward and inverse kinematics, inv dynamics etc.
- For the iCub (~30-50dof (?)), same but also getting the mass matrix, COM etc.

+ add a mid-range computer results in the readme and maybe raspberry results. Again, just to get an order of magnitude.
What do you think ?

traversaro, 2019-09-18:
It make sense, if it is something that is meant to be run by the users I would add it as a tool in https://github.com/robotology/idyntree/tree/master/src/tools , something like `idyntree-benchmark` that takes in input a model file, and if no model is specified runs the benchmark for two embeedded models.

traversaro, 2019-10-15:
Related repo: https://github.com/rbd-benchmarks/rbd-benchmarks .

closed 2023-09-10 by ahoarau

traversaro, 2023-09-10:
Interestingly, I started doing some benchmarks of iDynTree vs Pinocchio in https://github.com/ami-iit/idynfor/pull/39 and https://github.com/ami-iit/idynfor/pull/37 .