gbionics/idyntree #584: iDynTree build (Xcode generator) FAILS with option USE_ASSIMP=ON

nunoguedelha, 2019-09-27:


traversaro, 2019-09-27:
This could be related to https://github.com/assimp/assimp/issues/1517 .

lrapetti, 2019-10-01:
The same problem is happening on MacOS compiling with `cmake` and the error is the following:
```
[ 68%] Built target idyntree-estimation
[ 68%] Linking CXX shared library ../../lib/libidyntree-solid-shapes.dylib
ld: library not found for -lassimp
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [lib/libidyntree-solid-shapes.dylib] Error 1
make[1]: *** [src/solid-shapes/CMakeFiles/idyntree-solid-shapes.dir/all] Error 2
make: *** [all] Error 2
```

`assimp` has been installed trough `homebrew` and is at the latest stable head (`stable 4.1.0`).
I'm changing the title to reflect the problem is related to MacOS and not only to `Xcode` compiler.

closed 2020-04-02 by traversaro