gbionics/idyntree #590: Problem with loading RPY orientation data from URDF

MiladShafiee, 2019-10-06:
During using IMU data for push recovery, I realized that it seems there is some problem during loading frame orientation of head IMU from URDF model. I'm not sure that I'm doing a stupid mistake or there is a bug. As @traversaro suggested I prepared a simple main file that just loads the model and print data. For example I use data of "imu_frame" in the URDF file that is 
rpy="-1.57079632679 1.57079632679 0"/>

However the data that I find in the code is:
[INFO]head imu Roll 0 
[INFO]head imu Pitch 1.5708 
[INFO]head imu Yaw 1.5708 

The repository for reproducing this problem:
https://github.com/MiladShafiee/TestiDynTree

traversaro, 2019-10-06:
Hi @MiladShafiee , thanks for providing an example of the issue that you described . However, you should provide some information about the code, for example how to compile it and how to run the code. In particular, in that repo you committed also the build files of an in-source build, and so it is not possible to compile the code by just running: 
~~~
git clone 
~~~
To actually compile the code, I need to remove the `CMakeCache.txt`, and then I am able to compile the code. However, note that both using the `in-source builds` and commiting the build files to a repo are both things that are strongly discouraged, see https://cgold.readthedocs.io/en/latest/tutorials/out-of-source.html and http://alblue.bandlem.com/2011/03/git-tip-of-week-ignoring-build-output.html  . 

After I cancel the `CMakeCache.txt` (and modify the `CMakeLists.txt` to be actually able to compiile) if I try to run the executable I get this error: 
~~~
straversaro@iiticublap103:~/src/TestiDynTree/build$ ./debug_RPY_iDyn 
||| did not find model.urdf
[INFO][WalkingModule::setRobotModel] The model is found in:   
I/O warning : failed to load external entity ""
[ERROR] ModelLoader :: loadModelFromFile : Error in parsing model from URDF.
[ERROR][WalkingModule::setRobotModel] Error while loading the model from  
~~~

I guess that I need to specify manually the `--model` location, and so I do so, but it would be polite to you to specify this submitting the source code. : )

traversaro, 2019-10-06:
So, everything seems to be ok, as if you print the rotation corresponding to the RPY of `-1.57079632679 1.57079632679 0` and `0 1.57079632679 1.57079632679`: 
~~~cxx
std::cerr << iDynTree::Rotation::RPY(-1.57079632679, 1.57079632679, 0.0).toString() << std::endl;     
std::cerr << iDynTree::Rotation::RPY(0.0, 1.57079632679, 1.57079632679).toString() << std::endl;     
~~~
in both cases, you obtain (ignoring some negligible numerical error): 
~~~
 0 -1 0
 0  0 1
-1  0 0
~~~
this is because for this particular rotation, infinite RPY representation correspond to the same orientation. In a nutshell, what you are seeing is completely normal.

traversaro, 2019-10-06:
Small note that does not change the above reply, but it is important: in your code what you are printing is not the `head_R_imu_frame` transform, but the `chest_R_imu_frame` transform, as you are not considering the `neck` joints in your model, and so the `head` link is merged in the `chest` link, and so the `imu_frame` has become an additional frame of the `chest` frame. This does not change the issue as I guess that when the neck joints are `0.0 0.0 0.0` the rotation between the `head` and the `chest` is the identity, but it is important anyhow to take this in consideration.

traversaro, 2019-10-06:
@MiladShafiee as far as I see everything is working as expected, so no issue need to be fixed in the code. If you agree as well given by replies, feel free to close the issue, thanks!

MiladShafiee, 2019-10-06:
Hi @traversaro , Thank you very much for all information and help and sorry for disturbing.

> I guess that I need to specify manually the --model location (as in ./debug_RPY_iDyn --model ../app/robots/iCubGazeboV2_5/model.urdf)

I don't know why It causes the problem, I used exactly this project and It automatically find the model but I run the project with QT, for building the project I used: 

> ccmake .
> cmake . 
> make 
> make install

> this is because for this particular rotation, infinite RPY representation correspond to the same orientation.

I checked this with the [this website](https://www.andre-gaschler.com/rotationconverter/):
For example for the `(0,1.57,1,57)` the rotation matrix is:
```
[  0.0000000,  0.0000037,  1.0000000;
   1.0000000, -0.0000037,  0.0000000;
   0.0000037,  1.0000000, -0.0000037 ]
```
But for  `(-1.57079632679 1.57079632679 0)` returns:
```
[ -0.0000037, -0.0000000,  1.0000000;
  -1.0000000, -0.0000037, -0.0000037;
   0.0000037, -1.0000000,  0.0000000 ]
```

I guess this problem is because of using different type of Euler angles in iDynTree and this website.

> as you are not considering the neck joints in your model, and so the head link is merged in the chest link, and so the imu_frame has become an additional frame of the chest frame.

Interesting point! Thank you very much for your help and time.

closed 2019-10-06 by MiladShafiee

traversaro, 2019-10-06:
> I guess this problem is because of using different type of Euler angles in iDynTree and this website.

That website provides several convention for Euler angles, if you select `ZYX` you get the type of RPY angles used in several robotics software projects, including iDynTree, URDF and the OPC UA Robotics Information Model.

traversaro, 2019-10-06:
> > I guess that I need to specify manually the --model location (as in ./debug_RPY_iDyn --model ../app/robots/iCubGazeboV2_5/model.urdf)
>
> I don't know why It causes the problem, I used exactly this project and It automatically find the model but I run the project with QT, 

Are you familiar with how the YARP's ResourceFinder finds files? See https://www.yarp.it/yarp_resource_finder_tutorials.html for more details. I suspect that in your system the `model.urdf` file is found somewhere in the system, probably due to some setting of the `YARP_DATA_DIRS` env variable. 

> for building the project I used:
> 
> > ccmake .
> > cmake .
> > make
> > make install
> 

That is a in-source build, that is strongly suggest to avoid for multiple reasons (ask f2f to anyone in the lab for more details). We typically create a separate build directory, and install in a local directory as well, for example: 
~~~
mkdir build
cd build
cmake -DCMAKE_INSTALL_PREFIX=<prefix> ..
make
make install
~~~

MiladShafiee, 2019-10-07:
> Are you familiar with how the YARP's ResourceFinder finds files? See https://www.yarp.it/yarp_resource_finder_tutorials.html for more details. I suspect that in your system the model.urdf file is found somewhere in the system, probably due to some setting of the YARP_DATA_DIRS env variable.

No I'm not familiar thanks for clarification. Yes in the bashrc there is `export YARP_ROBOT_NAME="iCubGazeboV2_5",`

> >for building the project I used:
ccmake .
cmake .
make
make install
> That is a in-source build, that is strongly suggest to avoid for multiple reasons (ask f2f to anyone in the lab for more details). We typically create a separate build directory, and install in a local directory as well, for example:

Thanks , Yes I already have understood your point  :wink:  I just wrote again the build steps that I did before, to explain for you that maybe the reason of loading the URDF file in my computer was because of my process of building.
Thanks again for the clarification.