gbionics/idyntree #593: iDynTree::Regressors, include documentation for writing regressor structure generally, and for floating base.

mandralis, 2019-10-09:
Dear authors, 

Would it be possible to give an example of writing the Regressor XML file for a floating base robot ? I would like to use the method `loadRegressorStructureFromString(regrXml)` but I am unsure how this should be specified. 

In the python examples folder of the iDynTree repository, you give an example for the iCubLeftArmFTRegressor, where the following code is given:

``` 
regrXml =  '<regressor>' 
           '  <subtreeBaseDynamics>' 
           '   <FTSensorLink>r_upper_arm</FTSensorLink>' 
           '   </subtreeBaseDynamics>' 
           '</regressor>'
``` 
**First of all**: what do the tags `<subtreeBaseDynamics>` and `<FTSensorLink>` mean ? 
**Secondly**: what do I need to modify in order to specify the structure of the regressor for a floating base robot (for example a quadruped).
**Thirdly**: why do we need to specify the structure of the regressor in the first place, and what should it contain in order to return an accurate regressor for the robot specified in the URDF file ?

Thank you.

mandralis, 2019-10-10:
Thank you @traversaro. Is there any documentation for the classical inverse dynamics regressor i.e. how to use that method? Furthermore, where can I read about the idea of why you calculate the regressor in a certain way ?

mandralis, 2019-10-11:
Another related question @traversaro. You use inverse dynamics to calculate the regressor. However, this only works if the equations of motion are already linear with respect to the inertial parameters, right ? How do you guarantee this in your code base ? 

If I want to use this type of calculation in another code base what do I have to do to guarantee that the equations will be linear with respect to the inertial parameters ? Is it enough to use inertia tensors around the joint frame as shown in the paper by Atkeson et. al. ? How do you deal with the floating base ?

traversaro, 2020-01-08:
Sorry for the late reply @mandralis (feel free to ping in any issue if I forgot to reply : )  ). 

> Thank you @traversaro. Is there any documentation for the classical inverse dynamics regressor i.e. how to use that method?

Which method do you refer with "that method" case? If you refer to `inverseDynamicsInertialParametersRegressor` , the precised definition is indeed given in equation 7.1 of my PhD thesis, see https://traversaro.github.io/phd-thesis/traversaro-phd-thesis.pdf . 
For some related literature on how to use it, you can check the following papers: 
* Identifiability and identification of inertial parameters using the underactuated base-link dynamics for legged multibody systems https://journals.sagepub.com/doi/abs/10.1177/0278364913495932 : A nice paper that explain in detail the different between the **fixed base regressor** that is tipically used for industrial robots, and the floating base regressor that you use for floating base robots
*  Geometric Robot Dynamic Identification: A Convex Programming Approach  - https://ieeexplore.ieee.org/document/8922724 : By far the best model based identification algorithm for inertial parameters, the key point is the use of a regularization term that is not the normal euclidean one

> Furthermore, where can I read about the idea of why you calculate the regressor in a certain way ?

What you mean by "why you calculate the regressor in a certain way" ?

> You use inverse dynamics to calculate the regressor. However, this only works if the equations of motion are already linear with respect to the inertial parameters, right ? How do you guarantee this [...]

You can always write the inverse dynamics linearly with the inertial paramers, if you properly define the mapping between the density of the body and the ℝ¹⁰ . As discussed in the paper you mentioned, the tipical trick to obtain that is to define the inertia matrix in a frame that is not placed in the center of mass. This is a discussed a bit also in https://ieeexplore.ieee.org/abstract/document/7759801, let me know if it is understanding. 

> How do you deal with the floating base ?

I think the paper "Identifiability and identification of inertial parameters using the underactuated base-link dynamics for legged multibody systems" mentioned before could be helpful w.r.t. to that.

traversaro, 2021-09-15:
Feel free to re-open or open new issues if you think it is useful.

closed 2021-09-15 by traversaro