gbionics/idyntree #595: Define operator[] for Vectors

GiulioRomualdi, 2019-11-09:
In order to access element of an iDynTree vector the `operator()` has to be used. Having the `operator[]` may be useful when template functions are implemented. I.e. `std:: vector<>` uses the `operator[]` to provide the access to the elements.

I noticed that this strategy is already implemented in `Eigen` (check [here](https://eigen.tuxfamily.org/dox/classEigen_1_1DenseCoeffsBase_3_01Derived_00_01ReadOnlyAccessors_01_4.html#a496672306836589fa04a6ab33cb0cf2a) for further details)

traversaro, 2019-11-09:
I totally agree. I think the original rationale for using `operator()` was to provide a consistent API with  Matlab and Python (or perhaps it was Lua), in which the `operator[]` could not be overloaded. However, I do not think it is a problem to overload both `operator()` and `operator[]`.

traversaro, 2019-11-19:
Fixed in https://github.com/robotology/idyntree/pull/596 .

closed 2019-11-19 by traversaro