gbionics/idyntree #599: iDynTree cmake fails with assimp5 in MacOS (DEVEL)

lrapetti, 2019-11-20:
Recently, the bottle of `assimp` has been moved to `assimp5` (https://github.com/Homebrew/homebrew-core/commit/d4ccd465d36df85d0caa19e3974d88aaf618f8d7), and CMake is failing with the following error:
```
CMake Error at /usr/local/lib/cmake/assimp-5.0/assimpTargets.cmake:85 (message):
  The imported target "assimp::assimp" references the file

     "/usr/local/lib/libassimp.dylib.5"

  but this file does not exist.  Possible reasons include:

  * The file was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and contained

     "/usr/local/lib/cmake/assimp-5.0/assimpTargets.cmake"

  but not all the files it references.
```

Temporary fix was installing previous version of assimp
```
brew uninstall assimp
brew install https://raw.githubusercontent.com/Homebrew/homebrew-core/ca08d3c78a9a8f4e6a5249389442ebcb962eea0d/Formula/assimp.rb
```

lrapetti, 2019-11-20:
Important note, the cmake failure is happening even if ` IDYNTREE_USES_ASSIMP` is set to **`OFF`**

traversaro, 2019-11-20:
> Important note, the cmake failure is happening even if ` IDYNTREE_USES_ASSIMP` is set to **`OFF`** and `assimp5` is installed

Yes, unfortunately CMake gives error if a package has a malformed `<package>Targets.cmake`, even if `find_package(<package>)` is invoked without `REQUIRED` . For the time being, you  can comment out the line `idyntree_handle_dependency(ASSIMP)` in https://github.com/robotology/idyntree/blob/devel/cmake/iDynTreeDependencies.cmake#L60 . However, this is a Homebrew  problem, did you already searched if there is any issue on this in homebrew?

traversaro, 2019-11-20:
What is the name of  correct assimp lib installed by homebrew?

traversaro, 2019-11-20:
Ok, problem reproduced in https://github.com/robotology/idyntree/runs/311505150#step:10:44 . As said before, the problem is actually in the assimp bottle provided in homebrew, so I would strongly suggest to open an issue in https://github.com/Homebrew/homebrew-core/issues , probably first understanding what is the correct name of the library. You can also try to manually edit the `/usr/local/lib/cmake/assimp-5.0/assimpTargets.cmake` file to reference to the actually library name.

traversaro, 2019-11-26:
Sorry, due to https://github.com/robotology/robotology-superbuild/pull/306 I did not realized that `assimp` is a transitive dependency of Gazebo, and so it is **always** installed if a user follows the instructions available at https://github.com/robotology/robotology-superbuild . @lrapetti how did you handled this in your system?

traversaro, 2019-11-26:
I tried to open an issue in https://github.com/Homebrew/homebrew-core/issues/new?template=bug.md , but unfortunatly not having a macOS machine, it is tricky for me to provide them all all the debug info they request, if some macOS users could open an issue for this it would be great. 
@lrapetti @kouroshD @nunoguedelha @CarlottaSartore @claudia-lat

lrapetti, 2019-11-26:
> @lrapetti how did you handled this in your system?

For the moment the only fix I have found was to:
- force uninstall assimp
  `brew uninstall --ignore-dependencies assimp`
- compile `iDynTree` setting `USE_ASSIMP` to `OFF`
- reinstall assimp
  `brew install assimp`

lrapetti, 2019-11-26:
> if some macOS users could open an issue for this it would be great.

I have opened an issue in `homebrew-core` (https://github.com/Homebrew/homebrew-core/issues/47210), feel free to add any information to the issue.

lrapetti, 2019-11-26:
Anyway I have just found a related issue (https://github.com/assimp/assimp/issues/2760) with the associated fix (https://github.com/assimp/assimp/pull/2765)

lrapetti, 2019-11-27:
> > if some macOS users could open an issue for this it would be great.
> 
> I have opened an issue in `homebrew-core` ([Homebrew/homebrew-core#47210](https://github.com/Homebrew/homebrew-core/issues/47210)), feel free to add any information to the issue.

The issue in `homebrew-core` has been closed, we should wait the bug to be fixed upstream and then the bottle to be updated. Anyway, for the moment https://github.com/robotology/idyntree/pull/600 prevents the problem when `USE_ASSIMP` is OFF

traversaro, 2019-11-27:
> The issue in `homebrew-core` has been closed, we should wait the bug to be fixed upstream and then the bottle to be updated. Anyway, for the moment #600 prevents the problem when `USE_ASSIMP` is OFF

Interesting, it seems that bottles for dartsim were created correctly using assimp5 (see https://github.com/osrf/homebrew-simulation/pull/867), probably because dart ships its own `FindAssimp.cmake` file, see https://github.com/dartsim/dart/blob/master/cmake/DARTFindassimp.cmake .

lrapetti, 2019-12-02:
It seems the patch is now merged on a release upstream branch https://github.com/assimp/assimp/pull/2765

lrapetti, 2020-01-10:
Homebrew formula for assimp has now been updated with the patch, so the problem should now be fixed

traversaro, 2020-01-10:
> 
> 
> Homebrew formula for assimp has now been updated with the patch, so the problem should now be fixed

Related Homebrew's PR : https://github.com/Homebrew/homebrew-core/pull/47792 .

Let's test if this is working with https://github.com/robotology/idyntree/pull/623 .

traversaro, 2020-01-10:
> Let's test if this is working with #623 .

Apparently it is not working, perhaps the bottles still needs to be updated after https://github.com/Homebrew/homebrew-core/pull/47792 ? I tried checking the history of the Formula but GitHub is unable to display the history, perhaps checking locally could be  give you more info.

lrapetti, 2020-01-15:
The bottle has been further [updated to `assimp5.0.1`](https://github.com/Homebrew/homebrew-core/commit/f5e9560b6f6b70f9a14582fa7b251b73b994c6f8), I tested it and I am no longer experiencing the error described in this issue.

However, I am know getting the following compilation error when `USE_ASSIMP` is `ON`:
```
[ 98%] Linking CXX shared library ../../lib/libidyntree-solid-shapes.dylib
ld: library not found for -lassimp.5
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [lib/libidyntree-solid-shapes.dylib] Error 1
make[1]: *** [src/solid-shapes/CMakeFiles/idyntree-solid-shapes.dir/all] Error 2
make: *** [all] Error 2
```

@traversaro could you please trigger te test of #623 so at least we understand if the CMake error is solved, then eventually we can open a new issue for the new failure.

traversaro, 2020-01-17:
Done, the error is indeed that one: https://github.com/robotology/idyntree/pull/623/checks?check_run_id=391750053 .

lrapetti, 2020-01-17:
I change the title accordingly, since the problem is no longer related to `CMake`

traversaro, 2020-01-17:
> 
> 
> I change the title accordingly, since the problem is no longer related to `CMake`

More and less, I guess the problem it still in the CMake config files of assimp that do not contain the correct library name.

traversaro, 2020-04-01:
With we made a bit of progress with https://github.com/robotology/idyntree/pull/661 , the compilation now fails with the following message (see https://github.com/robotology/idyntree/runs/552452446): 
~~~
 [ 71%] Building CXX object src/solid-shapes/CMakeFiles/idyntree-solid-shapes.dir/src/InertialParametersSolidShapesHelpers.cpp.o
/Users/runner/runners/2.165.2/work/idyntree/idyntree/src/solid-shapes/src/InertialParametersSolidShapesHelpers.cpp:22:10: fatal error: 'assimp/Importer.hpp' file not found
#include <assimp/Importer.hpp>
         ^~~~~~~~~~~~~~~~~~~~~
1 error generated.
make[2]: *** [src/solid-shapes/CMakeFiles/idyntree-solid-shapes.dir/src/InertialParametersSolidShapesHelpers.cpp.o] Error 1
make[1]: *** [src/solid-shapes/CMakeFiles/idyntree-solid-shapes.dir/all] Error 2
~~~
If one of the macOS users wants to provide the `assimp-*.cmake` files present in their homebrew installation, probably that can help the debug.

closed 2020-04-02 by traversaro