gbionics/idyntree #603: Implement an extractor of OPW parameters from iDynTree::Model

traversaro, 2019-12-07:
As we have an extractor of DH parameters from iDynTree::Model, it would be nice to have a tool that would be able to extract OPW parameters (see their definition in https://github.com/Jmeyer1292/opw_kinematics) from 6-dofs iDynTree::Model . 
As in the DH case, the converter should give an error if the input model is not respecting the assumption provided in https://github.com/Jmeyer1292/opw_kinematics, in particular if the model does not have 6 dofs.

ahoarau, 2019-12-07:
almost no URDF is built with the DH representation. To me this extractor has to make sure to find an equivalent DH representation for a given URDF. That's a cool intership ;)

traversaro, 2020-01-30:
>  To me this extractor has to make sure to find an equivalent DH representation for a given URDF. That's a cool intership ;)

I think we already have this, check the `urdf2dh` tool in  https://github.com/robotology/idyntree/blob/master/src/tools/urdf2dh.cpp .

traversaro, 2020-01-30:
Note that `urdf2dh` tools depends on YARP and ICUB, but that is just to output a YARP compatible `.ini` file, the core of the method is already part of https://github.com/robotology/idyntree/blob/master/src/model/include/iDynTree/Model/DenavitHartenberg.h and it has no dependencies (expect perhaps Eigen3). Creating a standalone urdf2dh command that just prints out the DH parameters or converts them to json format should be trivial. Now that I think of, these small utility executables are stuff that it could make sense to just distribute as statically linked executable for Win/Mac/Linux/Wasm to simplify their use.